serkri/SmartEVSE-3 #170: Mqtt: home battery set to 0

stevoh6, 2023-07-13:
Hi, im feeding SmartEvse with data via mqtt. I see strange behaviour, when I send MainsMeter data, home battery is set to 0 until I resend HomeBatteryCurrent data again. Those data are comming in diferrent intervals, because they are now separated. In API we send those data together.

dingo35, 2023-07-14:
Ok I don't have a home-battery, and I don't feed MainsMeter currents through the API, so I'm not talking from practical experience here; but we never documented the homebatterycurrent API being valid for Solar Mode only. Now Normal Mode doesnt make any sense, but wouldn't Smart Mode be an understandable use case?

So I have 25A MainsConnection, and a 5A homebattery feed, so now I would like to charge 30A in Smart Mode?

arpiecodes, 2023-07-14:
I also have no idea to be honest. @stevoh6 what's your use-case by using the battery current?

stevoh6, 2023-07-14:
Use case is simple, solar panels + home battery + solar mode on evse.  I was cecking/observing flowing data in mqtt server and I notice, that battery is until "battery update is send" reset to 0 after evse receive currents. I will check it more precicely nce again. But I have two problems with home battery, second one is major:
1. > I would like to see battery info even Im on normal mode on web ( so  I doesnt need to check another web page to get this info in order set override currents) -not critical.
2. > when Im in solar mode, the changes in currents are too frequent. Im feeding evse via mqtt with currents every (5-10s > when values are updated in HA >from inverter). Smart evse is trying agrresively adjust currents and every 5s the energy flow in my house is crazy, first 5 seconds Im importing from grid, second 5 seconds im importing from battery, third 5s im importing from grid and so on. I woul like to stabilize that, for example adujt it every 1-5mins, ignore some small values like +-200w,  Most of us are fine, when for short period of time ( clouds) we feed car charging from home battery. 

I was trying to sme crazy automations to adjust "battery data before sending them to evse, but without any feasible luck.
```
alias: .EVSE set MQTT HomeBatteryCurrent from HA
trigger:
  - platform: state
    entity_id:
      - sensor.battery_current
    for:
      hours: 0
      minutes: 0
      seconds: 1
action:
  - service: mqtt.publish
    data_template:
      topic: SmartEVSE-15780/Set/HomeBatteryCurrent
      payload: >-
        {% set battery_current10 = states('sensor.battery_current') | float * 10
        * (-1) %} {% set charging = iif(battery_current10 > -10, true , false)
        %} {% set over_charging = iif((30 > battery_current10 ) , false , true)
        %} {{ iif(charging, iif(over_charging, battery_current10 - 30, 0 ),
        battery_current10 - 10 ) }}
```

dingo35, 2023-07-14:
Why would you want to stabilize your currents? Electricity usage in your house is per definition changing every micosecond, every switch you flip, every sensor in your washing machine or coffee maker, every cloud over your solar panels.... it doesnt make sense?!

arpiecodes, 2023-07-14:
I agree with @dingo35, it'd be much easier to dynamically set CurrentOverrides on SmartEVSE based on a certain baseline you determine with Home Assistant. Also, wouldn't it make much more sense to use HA as your central dashboard containing all of the information vs. the web interface of the SmartEVSE?

You explain the solution, but not the actual use case. It's still not clear why you want to do what you currently do. :)

stevoh6, 2023-07-14:
yes HA is main dashboard for everithing. Basicaly if I have 8kW solar power, house has consumption 0.5kW  I woul like to charge car with 7kW and keep 0.5kW as buffer to not switching between grid import/grid export,  home battery charge/discharge.... I know that it is changing constantly, but  it isnt better to just change current of grid export insted of switching gbetween grid import/export? Maybe Im over-complicating it :D

arpiecodes, 2023-07-14:
Aren't you then better off placing an PV meter in between, and put SmartEVSE in solar mode so it only will charge when you are producing solar power? I mean sure, there's also a chance your battery still has to recharge, but you have other means to control whether EVSE will allow charging or not (for example; toggle the access property on/off based on your SoC of the battery by automation in HA).

closed 2023-08-04 by dingo35