serkri/SmartEVSE-3 #179: Enable to put system in OFF mode or otherwise force disconnect from buttons on front panel

Ivesvdf, 2023-07-15:
In SERKRI-1.7.0 it seems impossible to force ongoing charging to stop using the front-panel SmartEVSE integrated buttons. Previously I think it was possible to set the mode to OFF which would force a disconnect however now changing mode to OFF while connected and charging does not seem possible. In my setup only NORMAL mode can be selected in the menu while charging.

This would be useful when wifi isn't working and if a disconnect switch has not been fitted. Currently in this scenario you have to rely on the vehicle to disconnect.

dingo35, 2023-07-15:
I am unaware of this "feature" (bug?).

How in the past did you interrupt charging with the LCD screen/buttons? What version firmware did you use? Do you have RFID reader and/or switches installed?

Ivesvdf, 2023-07-15:
I don't have an RFID reader or button installed. 

Looking into it more I guess it was never possible to do this cleanly - I think I probably changed the mode to SOLAR which causes a fault because I don't have a mainsmeter which in turn stops charging. So definitely not ideal. 

Perhaps I should phrase it as a feature request more than a bug in that case. For example. while in non-settings mode the rightmost button or leftmost could be used identical to how the SmartEVSE Power toggle in the Home Assistant add-in functions (ie switch between off and normal if no mainsmeter). Probably also with a 2 second delay like for opening the menu.

arpiecodes, 2023-07-15:
Sounds like a useful feature to have. E.g. toggle access by long-pressing a certain button or both arrow buttons at the same time.

dingo35, 2023-07-27:
[startstop.zip](https://github.com/serkri/SmartEVSE-3/files/12181768/startstop.zip)
Please test:

# Improved starting/stopping through the LCD screen
* When pressing o button longer then 2 seconds you will enter the Menu screen
* When pressing < button longer then 2 seconds the access will be denied, i.e. the mode will be set to "Off" and charging will stop
* When pressing > button longer then 2 seconds the access will be granted, i.e. the previously set mode will be activated and charging will start

Ivesvdf, 2023-07-27:
Looks great, I have today off so I was able to test "09:48:35 @Jul 27 2023" pretty much immediately. 

- Tested both start and stop through buttons while no vehicle connected -> OK 
- Confirmed menu button still works -> OK 
- Tested both start and stop with vehicle connected, charging stops -> OK 
- Confirmed charging cannot start while turned off through buttons -> OK

So yeah no remarks at all, looks good!

dingo35, 2023-07-27:
Commit 4ac5bfb7

closed 2023-07-27 by dingo35