serkri/SmartEVSE-3 #182: Timezone problem ?

deekoowee, 2023-07-19:
Have to count 1 hour difference to start and stop setting. 
Starting and ending times have to set one hour less than real/wanted start/stop time.
[image]

dingo35, 2023-07-19:
Go to the LCD screen and setup your Wifi again; notice on the portal screen it should mention an autodetected timezone.

After completing your wifi setup all should be ok.

If not, please post the timezone mentioned on the portal screen, and your actual timezone.

deekoowee, 2023-07-20:
Made the Wifi setup, but there's no timezone info at portal
The time difference at portal changed from 1 hour to 3 hours
Timezone is EEST, +0300
1.7.1 is in use

dingo35, 2023-07-20:
Look at the very first screen, if I remember correctly it is where the "config" button is. Down under it displays something likr: Timezone: Europe/Berlin.

deekoowee, 2023-07-28:
I can't see the timezone
[image]

Here's the screenshot from worldtimeapi
[image]

dingo35, 2023-07-28:
You are misunderstanding me about the portal:
1 go to your LCD screen
2 choose Menu/Wifi/Setup; it says something like "move up to start portal"
3 move up
4 now the portal starts on an access point with name "SmartEVSE-xxxxx"
5 tune your smartphone to this wifi access point
6 after wifi connection is estsblished, surf to 192.168.4.1 on your smartphone
7 you see a screen with "config" button; on this screen your detected timezone is displayed; I need a screenshot of this screen
8 press config and finish your wifi setup; this should save your timezone to your flash memory correctly

Then check if the problem is solved, and upload your screenshot here.

I hope you understand it is impossible for me to test your situation , since I should have to move to your timezone.....

dingo35, 2023-07-28:
[image]

deekoowee, 2023-07-28:
[image]
I re-setup the wifi and tried set the smart charging, but it is having the same 3hours difference.
Set 17:00 to 17:10, StartTime is 21:00 and StopTime is 21:10

dingo35, 2023-07-28:
What is this picture a screenshot from? It does not look like the portal ?!?

deekoowee, 2023-07-28:
Yes, from information page.
Times were my mistake, I didn't wait enough before set the time again.
3 hours is correct difference.
Difference seems to be UTC+3 as reported from worldtimeapi

dingo35, 2023-07-29:
Ok, I made a special version of the firmware where I hardcoded timezone = "Europe/Tallinn"; on the LCD screen then the Tallinn time is shown, but on the webserver it still showed my (CEST) time.
Then I changed the settings of my operating system, set the time zone to Tallinn also, and now also the webserver shows me the Tallinn time, but also the settings work correct:
[image]

You see when I input 10:50 as a starttime it displays it correctly.

I include the "hardcoded" Tallinn version here, could you see if this solves your problem?
[tallinn.zip](https://github.com/serkri/SmartEVSE-3/files/12206600/tallinn.zip)
Please flash both firmware.bin AND spiffs.bin !!!

dingo35, 2023-07-29:
Just to be sure: your operating system /browser have timezone set to Tallinn, right?

deekoowee, 2023-07-30:
Time settings in windows;
[image]
Daylight savings is checked and system time is correct.

dingo35, 2023-08-01:
Ok well hardcoding the timezone is obviously not the problem, I need you to work with me to solve this for all future versions, in all timezones. Please follow my instructions closely!
[tallinn2.zip](https://github.com/serkri/SmartEVSE-3/files/12226138/tallinn2.zip)

Please flash tallinn2.zip, both firmware.bin and spiffs.bin. DO NOT go into the wifi-setup-portal!! Now when you go into the normal webserver, press the RAW button; the last 5 items in the "settings"  chapter are all about timezone: TZread, TZwritten, TZset, timezone, and tz_info. Please post all 5 values in this thread.

Now please go into the wifi-setup-portal, go through the wifi-setup procedure so it completes succesfully, and then again go to the normal webserver, press RAW, and post again the 5 timezone items.

Hopefully we find the problem!

deekoowee, 2023-08-01:
Updated the controller with tallinn2.zip
Copied the following data from settings page source.

Settings data before WiFi setup;

"TZread":"Europe/Tallinn"
"TZwritten":""
"TZset":"EET-2EEST,M3.5.0/3,M10.5.0/4"
"timezone":"Europe/Tallinn"
"tz_info":"EET-2EEST,M3.5.0/3,M10.5.0/4"}

Settings data after WiFi setup;

"TZread":"Europe/Tallinn"
"TZwritten":"Europe/Tallinn"
"TZset":"EET-2EEST,M3.5.0/3,M10.5.0/4"
"timezone":"Europe/Tallinn"
"tz_info":"EET-2EEST,M3.5.0/3,M10.5.0/4"}

TZwritten have now the timezone info

dingo35, 2023-08-01:
Thanks for your debug report. In this new version, does the problem still exist or is it solved now?

deekoowee, 2023-08-02:
Yes, tallinn2 works ok and problem is solved. Thanks.

closed 2023-08-02 by dingo35