serkri/SmartEVSE-3 #183: Control via IoBroker using API

Elektrofix-OL, 2023-07-21:
I am trying to control the SmartEVSE with the IoBroker. Since the SunnyHomemanger is not directly supported, I would like to send the data from the eHZ to the EVSE. I'm trying a url call 
http://192.168.5.23/currents?L1=15&L2=-10&L3=-11
but the EVSE shows nothing in the browser. In the EVSE I set the MainMeter to API.

What am I doing wrong?

dingo35, 2023-07-21:
You should send values in Amps x 10; so if you want to send 1.5A you should send 15.

You could try using curl, like: curl -X POST http://etc. , if that works than your system is either sending a GET or not the right syntax...

Also note you should send the info every few seonds, and it takes the webserver UI 6secs to refresh

Elektrofix-OL, 2023-07-21:
I got it. It is important that it is not a „request“ but an „exec“ as curl with ' character, otherwise nothing arrives.
Here is my Blockly:
[image]

closed 2023-07-23 by Elektrofix-OL