serkri/SmartEVSE-3 #184: Charge cancellation in Smart and Solar mode

Elektrofix-OL, 2023-07-26:
In smart mode, charging stops after a while and won't start again unless I go to normal and then switch back to smart after starting. At
Solar the same phenomenon, although 14.5A go into the grid.
It doesn't matter whether I also transfer the current from the house battery to the API, it also aborts. After the cancellation, Sate says "Connect to EV

Anyone have an idea?

My structure:
- Mainmeter via API new data every 10 s
- 1-phase charging
- MQTT connection to IoBroker
- Charging current limited to 6A
- Car connected with 60% and able to take 1x16A/3x32A
[image]
yellow line shows the actual current, green shows the current specification of the EVSE

dingo35, 2023-07-26:
Does the LCD display show an error? Feeding the API every 10s is on the edge, better feed it every few seconds.

dingo35, 2023-07-26:
You sure you dont have a networking problem?
Network congestion on sending a few hundred bytes every few seconds on a >100Mbps link sounds strange....

dingo35, 2023-07-27:
Apart from your networking problems, please post a verbose telnet log starting some time before the end of the charging session, and ending after the end of the charging session; then we can see what is going on....

Elektrofix-OL, 2023-07-27:
Yes, I'm sure my network is OK. I have now logged mqtt SmarEVSE/Error. From time to time there is a communication error and after 1 sev. Everything is ok again.

What is a telnet protocol?

dingo35, 2023-07-27:
firmware.debug.bin has a telnet-server built in that you can use to log debug messages of the SmartEVSE. 
For details on telnet: google is your friend...

Elektrofix-OL, 2023-07-27:
Is that the desired type of info?

```
(loop)(C1) 0 clients running.
(I) (printStatus)(C1) STATE: A Error: 2 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C1) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C0) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C0) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C0) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C0) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.2 A IsetBalanced: 25.5 A
(I) (printStatus)(C0) L1: 1.2 A L2: -1.8 A L3: -1.6 A Isum: -2.2 A
(loop)(C1) 0 clients running.

```

dingo35, 2023-07-27:
No you need verbose  mode, press v and then enter

Elektrofix-OL, 2023-07-29:
Today I loaded the car again and logged the session with putty. Suddenly the connection ended and the car stopped charging. After 4.5 minutes the car started charging again.
[putty.log](https://github.com/serkri/SmartEVSE-3/files/12206493/putty.log)

dingo35, 2023-07-29:
The log just shows first no EV connected (state A), then EV gets connected and starts charging at 1x6A . So nothing strange there.

If it keeps happening you might increase your charge current to 7A, since 6A is exactly the minimum specified by the charging standards, and see if that changes things....

Elektrofix-OL, 2023-07-29:
I'll keep watching, I just need more sun, the weather here is currently quite cloudy.

I just don't understand why Putty disconnects for the second time, is it the EVSE?

dingo35, 2023-07-29:
No that seems to be a bug in the remotedebug library we are using....

closed 2023-07-31 by Elektrofix-OL