serkri/SmartEVSE-3 #192: MQTT: SmartEVSE publishing to strange topics and incorrect data

kuhlivisj, 2023-08-04:
I'm seeing some strange topics when monitoring MQTT. Also sometimes correct topics get strange values.
In the screenshot the lasst 4 topics are wrong and MainsCurrentL1 has a strange value.

[image]

My SmartEVSE is running v1.7.2.

dingo35, 2023-08-04:
If you watch the same data on the webserver, "Raw Data", same problem or values ok there?

kuhlivisj, 2023-08-04:
On the "Raw Data" page the values seem correct. 

On the "Raw Data" page it says that MQTT is disconnected yet it is publishing to the broker.

dingo35, 2023-08-04:
Does reconfiguring your MQTT settings on the webserver page change anything?

I cannot reproduce this at my site.

kuhlivisj, 2023-08-05:
I tried it. Changed the settings to some random value and saved them. The SmartEVSE indicated that it was disconnected.
Then I changed the settings back to the correct values and saved them. The SmartEVSE connected again and after a few minutes I see the same problem again.

It looks like it has something to do with writing the mainsmeter data to the SmartEVSE over MQTT. When I don't send it everything is working as expected. When I start sending the data again the problem is back.

dingo35, 2023-08-05:
Does v1.7.1 have the same problem?

kuhlivisj, 2023-08-05:
Tried v1.7.1 for 15 minutes now and it is working fine.

dingo35, 2023-08-05:
Are you willing and able to compile yourself?

dingo35, 2023-08-05:
It is not very complicated, and this way you can find out which commit causes the problem; first do 
git checkout (some commit in the middle of v1.7.1 and v1.7.2)    --> test if problem exists work your way down to 1.7.1 else work your way up to 1.7.2 .

Building the firmware

-     Install platformio-core https://docs.platformio.org/en/latest/core/installation/methods/index.html
-     Clone this github project, cd to the smartevse directory where platformio.ini is located
-     Compile firmware.bin: platformio run
-     Compile spiffs.bin: platformio run -t buildfs

kuhlivisj, 2023-08-05:
Commit 61f49cf1624630d816774078d07164c65477daff is causing the problems.

Tried a few different commits, but the problems start with that commit.

dingo35, 2023-08-05:
Interesting, that code updates publishes the currents immediately after they change; Im going to review that code tomorrow....

hmmbob, 2023-08-06:
It might not just be the currents though, my HA complains about some random data in other fields as well

[image]

dingo35, 2023-08-06:
@hmmbob Your problems also arise when you are feeding Mains Currents L1 - L3 via MQTT?
Also problems disappear when stop feeding?
Also problems disappear when using v1.7.1?

hmmbob, 2023-08-06:
I am not feeding L1-L3 through MQTT, but through API (http post). Rollback to 1.7.1 fixes the issue.

dingo35, 2023-08-06:
Im not, since I dont get to reproduce this problem.

If you want to check free ram, uncomment lines in src/evse.cpp that contain the string "free ram", and check debug log via telnet.

If necessary you can increase stack sizes of different tasks, src/evse.cpp look for comment "Stack size".

marcovaneck, 2023-08-07:
Rarely I see this problem. My 2 cents are in the multi-threading area. Since in 1.7.2 the Currents are updated in the _void UpdateCurrentData(void)_ method, which is called from several places. As a proof of concept I moved them back to the _mqttPublishData()_ method. Let you know if the problem disappears.

Ps. I push the current-data via the web-interface, not via MQTT.

dingo35, 2023-08-07:
I agree that is much more probable; and Im convinced the problem is solved by placing the publish statements back to the publishdata function.

But I dont see that as a solution, since the long term goal is to feed the mqtt server immediately (and only) if values change.

So the solution would be to make the publish routine threadsafe; to do that effectively it is necessary to find out what part of the library is the problem OR move to a threadsafe mqtt library.

dingo35, 2023-08-07:
[mutex.zip](https://github.com/serkri/SmartEVSE-3/files/12280834/mutex.zip)
As a proof of concept I placed mutex locks around the publish calls; could someone who can reproduce the problem test this?

kuhlivisj, 2023-08-07:
The version with the mutex is working fine. No more corrupted data and topics.

dingo35, 2023-08-08:
Fixed with 2020c53edcfb

closed 2023-08-08 by dingo35