serkri/SmartEVSE-3 #196: Mains info not showing over MQTT

hmmbob, 2023-08-08:
Running 1.7.3, the mains info seems to not get published for some reason. I am feeding the mains details over API (http post).

[image]

Mqtt explorer shows that the data isn't published to the broker:
[image]

dingo35, 2023-08-08:
Yep, look at the release notes of 1.7.3: currents fed via EM_API are no longer republished.

hmmbob, 2023-08-08:
Ah, I did not link that to this issue.

But... Why? I am http-posting them to the SmartEVSE, not to any other system. Now I am lacking this info, and particularly: I'd like to see what the evse is working with.

hmmbob, 2023-08-08:
In addition to the above question: the topics still get announced anyway (discovery).

But I actually hope that you want to reconsider the removal, I don't understand why it was removed in the first place?

dingo35, 2023-08-08:
I asked for a use case here  #192 and  nobody responded. 
So I would expect, once you transistion to MQTT, you would feed your SmartEVSE though MQTT too; and then your clients can subscribe to the source directly, without delays, and without burdening the SmartEVSE with relay tasks....

hmmbob, 2023-08-08:
Never saw your edit there, only the original message (email updates only send the original).

But publishing them over MQTT would mean that I need to add MQTT to the esp device plugged into the P1 port, burdening that device again. 

And the use case wouldn't be strictly relay by the way, it's simply keeping track of what the smartevse is using for its SMART mode calculations. But I guess I could do that on the device as well.

(Still wondering why the data _is_ being consumed by the HA integration then?)

hmmbob, 2023-08-08:
Oh, and the reason for feeding over http.post instead of MQTT: it's one link in the chains less. Http.post goes directly from P1 esp to SmartEVSE, for MQTT there's the broker in between (which impacts SmartEVSE workings if I restart/update the broker during charging for instance).

closed 2023-08-09 by dingo35