serkri/SmartEVSE-3 #197: Sensorbox v2 with clamps showing wrong values when powercable connected

VelocityMaximus, 2023-08-08:
**Issue:**
I am experiëncing a strange behavior with the sensorbox while using the powercable connected.

With the powercable disconnected it reads the values well within a few 10ths over the whole range that are measured with an external current  clamp and also P1 data.
The 3 clamps (100A model) are connected with the arrows towards the main fuse box. Powercable is connected to phase one.
It shows negative values when solar is sending power to the grid, assuming the setup is right.

The use case: One car is charging 3 phase, 1 car is charging 1 phase, the heatpump 3 phases was heating tap water and some other minor loads that were on phase 1.

**Scenarios tested:** 

- Phases are read well exept one depending on the load when the mains cable is connected.
- If there is solar power it looks right but the phases are a bit off (1 amp max difference between phases)
- When there is no solarpower and a slight usage of energy, phase 2 shows a negative current of -1.5A and phase 1 and 3 around 0,5A positive. In the HomeAssistant I read the P1 (4.2 version) it read that all phases get current from the grid.
- When there is no solar power, power cable connected and a high load ie: phase one 24 amps, phase 2 shows negative 7 amps and phase 3 around 6 positive.
- When I disconnect the powercable from the sensorbox it shows the right values: Phase 1 24,8A, phase 2 14,4A, phase 3 14,2A

**What can cause this issue?**
1. Currentclamps issue
2. Faulty sensorbox
3. Configuration
4. Calibration
5. Software issue
6. I am overlooking something

dingo35, 2023-08-08:
I would first flash the original firmware, to check whether this is software or hardware related.

Your last remark suggests something like ground/earthing loop problem....

dingo35, 2023-08-08:
Also power has to be fed from L1, as mentioned in the manual:

https://gathering.tweakers.net/forum/list_message/75736592#75736592

VelocityMaximus, 2023-08-08:
Phase one and clamp one are on the same wire. Other clamps connected to phase 2 and phase 3. 
Swapped sensors to see if the behavior changes but the values stay the same. Changed channel one clamp to another phase and then it goes way off the right values. 

Earthing is checked (I can not see what earthing does in the current lamp setup)

Do you mean the software from the sensorbox or the evse module?

dingo35, 2023-08-09:
I mean the firmware of the SmartEVSE box. AFAIK there is no alternative firmware for the sensorbox.

VelocityMaximus, 2023-08-09:
Just to be sure: Sensorbox plus and Sensorbox V2 act the same with CT clamps?

When I wire the sensorbox plus as it should be it measures wrong.
When I swap L1 and L3 clamps (CT1 is on L3 and CT3 is on L1) and leave the mainpower on L1 it reads the right values for power consumption and production.

The behavior of the above description is the same for 1.7.1 as per original 3.0.1

dingo35, 2023-08-09:
If it is not a firmware problem I recommend you check with the manufacturer; I have no sensorbox, so I never have tested my firmware with the sensorbox, and I try to stay away from the code that deals with it.

closed 2023-08-09 by dingo35

VelocityMaximus, 2023-08-09:
Addition:
I reconnected the clamps and have CT1 on L1 as per manual.
Then I connected CT2 to L3 and CT3 to L2.  This shows the right amount of current and direction except the values are swapped on the webpage and in HA. 

To me this looks like a phase swap thing in the software of the sensorbox.

I was reading the Sensorbox V2 software and see that there can be a phase rotation swap to run it CW or CCW. 
Maybe this is not possible with the Plus version. Maybe Michael can tell something about this. (also mailed him via the form in the webshop)

mstegen, 2023-08-16:
Sorry missed this.. 

It's very important that input 1 on the sensorbox is connected to the same phase as the power cable is connected to. 
This can also be L2 or L3! 
Make sure all arrows on the CT's point in the same direction.

Then the phase on connection 1 should measure the correct current. (this is displayed as L1)

If only L2 and L3 are swapped, then your phase rotation is CCW, you can force the Sensorbox+ and Sensorbox 2 to work in this mode, but i decided not to include it in the menu, as it's complicated enough already, and it might confuse people (only 3 wire/4 wire are selectable) 

For the SmartEVSE it doesn't really matter if they are swapped.