serkri/SmartEVSE-3 #199: Feature request: lock the menu access with code or SW enable switch (not accessible by menu)

the-master006, 2023-08-28:
Currently there is no way to lock the smartevse settings: if someone can access the smartevse controller, the settings can easily changed and someone can illegaly charge their car.
I see currently no way to prevent this. A pass code on the menu or something alike, or a software "enable" switch which is not accessible by the smartevse controller menu would fix this already.
Thanks!

dingo35, 2023-08-28:
AFAIK you can accomplish this with a switch and configure it to access B/S, where B = momentary switch and S = toggle switch (all this from Stegens original manual).

the-master006, 2023-08-28:
This does not work or works only partially: when the switch is pushed, then access denied shows on the display of the smart evse, but then still the menu is accessible by holding the middle button for 2 seconds, and then you can change the menu settings still, including the setting for the switch to be disabled or one of the other settings like you mention. 
So this is either a bug in the SW from Stegen, or extra functionality like I request.

dingo35, 2023-08-28:
Ok havent tested this, so it might work like you described.
So the requested change would be to disable the 2s timer that starts when one of the three buttons is started, when a switch is configured to access b or s?

dingo35, 2023-08-28:
[lockaccess.zip](https://github.com/serkri/SmartEVSE-3/files/12452682/lockaccess.zip)
Could you test this version _extensively_, with and without the switch configured to access, since I currently cannot test at my own config.

Thanks!

the-master006, 2023-08-28:
Allright, will test. Thanks!

negronoir, 2023-08-28:
I'll test also in the afternoon :-)

But this feature is interesting despite physical wired swith or 'sw switch' is used.

I mean, with original Stegen firware, I used the only possible way: a physical switch to disable access.

But with your firmware, I've removed wired switch and I set the HA control 'SmartEVSE Mode Selector' -> OFF to set 'SmarEVSE Access' -> FALSE

Controller menu will be locked also using HA 'sw sitch'?

Thanks!

negronoir, 2023-08-28:
Maybe I didn't explain myself ok @the-master006 :-)

The SW control already exists in Home Assistant and it is what I'm using instead a wired physical swith which I removed some time ago.

At the controller I've set that no switch in connected (Switch -> None)

But in HA, when yo set SmartEVSE Mode Selector (into the red box) to OFF value, SmartEVSE Access sensor (into the blue box) changes to FALSE and controller shows on the screen 'ACCESS DENIED' in the same way as a wired switch.

[image]

What I'd like is that despite of wired switch or HA SmartEVSE Access set to false, whaen controlles shows ACCESS DENIED, three buttons should be locked

Thanks!

dingo35, 2023-08-28:
I don't want to bother the majority of the users with software settings that are of no use to them.
If you made the (hardware) choice to hang your SmartEVSE out in public, IMHO you should also make the choice to block access by hardware.

negronoir, 2023-08-28:
I understand. Each particular situation is different.

I have my two SmartEVSEs in a neighborhood parking lot and I control them with HA from home.

I keep them into a box with a locker (the only way until today to avoid malicious people) but at least once I accidentally left the key unlocked and someone tried to change settings :-)

If it happens again I will put the wired switch back if you finally implement the locking feature

Thank you for all your work!!

dingo35, 2023-08-28:
So do you feed those SmartEVSE's with power from the public parking lot?
If feeding from home you could also consider cutting the power when not using it.

negronoir, 2023-08-28:
No! The neighbors would kill me LoL!

In Spain where I live, for installing a EVSE in the parking lot of an apartment building, a direct line from your electricity meter is installed. So a full protection box is needed in your EVSE equipment.

[image]

So no grid power control from my home. I've inside protection box a PLC to give WiFi to SmartEVSEs and the cars. If I use the PLC to power off the protection box, PLC also will be out :-)

I live in a safe town. No problems usually. I guess one of the days I left the door open, someone with the spirit of an engineer was playing haha

Thanks!

the-master006, 2023-08-28:
I have my smartevse just on my private driveway next to my house, but it is clearly visible from the street so it could be pretty easy for someone to just park the car on my driveway and start charging when I am not home.
So indeed what I would like that there is some software switch to get the access denied message including the disabling of the 3 buttons.
Because I could also enable the RFID functionality on the smartevse and only give access to my rfid tag/card, but then still the same holds I believe (not 100% sure although, I don't have a RFID reader currently): Anyone can still access the menu of the smartevse by the 3 buttons and disable the RFID reader functionality right? 
That seems not OK to me... Access denied should really be access denied, without any way to get access besides the true owner...
Thanks!

the-master006, 2023-08-28:
Hmm actually it now seems to work like above: when the access denied screen is shown, also the 3 buttons are disabled on the smartevse. And now with the smartEVSE Power option in HA you can easily switch between smartevse enabled and disabled, independant of the Mode selector.
Thanks!!

dingo35, 2023-08-29:
Honored by 2073d02802c53

closed 2023-08-29 by dingo35