serkri/SmartEVSE-3 #204: Slave SmartEVSe 1 won't start

rvdgaag, 2023-09-27:
I installed a second SmartEVSE, loadbalancing is on on the main unit, the secundary unit is node 1. The Main unit has an EV meter which reads fine on modbus nr 11

I also configured an EV meter on node 1 (as well as the main unit) on modbus nr 16 (Eastron 3 Phase SDM630v2).
The mains info is send over API to both the Master and Node1 (is this really nessesary ?)

The node 1 seems to see a car is connected but the car doesn't see it is connected obviously so it doesn't start charging.

Both SMARTEVSE's have the latest v1.7.3

Log file:

* Please type the command and press enter to execute.(? or h for this help)
***
(loop)(C1) 0 clients running.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0034.
v(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0034.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.8 A L2: -3.6 A L3: -2.5 A Isum: -9.9 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0034.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.6 A L3: -2.5 A Isum: -9.4 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.3 A L2: -3.6 A L3: -2.5 A Isum: -9.4 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.5 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.5 A Isum: -8.8 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0048.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(ModbusDecode)(C0) Modbus Exception 0, Address=16, Function=132.
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0034.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(MBhandleError)(C0) Error response: E4 - Server ID mismatch, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(loop)(C1) 0 clients running.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(MBhandleError)(C0) Error response: E2 - CRC check error, address: 10, function: 04, reg: 0006.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: -2.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: -3.3 A L2: -3.0 A L3: -2.2 A Isum: -8.5 A
[... 220 lines cut ...]
(ConfigureModbusMode)(C1) Setup MBserver/Node workers, end Master/Client
(ConfigureModbusMode)(C1) ConfigureModbusMode1 task free ram: 2324
(CheckAPpassword)(C1) APpassword: 08848079
(I) (write_settings)(C1) settings saved
(loop)(C1) 0 clients running.
(loop)(C1) 0 clients running.
(ProximityPin)(C0) PP pin: 1023 (1876 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
(I) (EVSEStates)(C1) Cable limit: 16A  Max: 16A
(setState)(C1) 15:15:07 STATE B1 -> COMM_B
(loop)(C1) 0 clients running.
(setState)(C0) 15:15:09 STATE COMM_B -> COMM_B_OK
(setState)(C1) 15:15:09 STATE COMM_B_OK -> B
(EVSEStates)(C0) Diode OK
(setState)(C0) 15:15:13 STATE B -> COMM_C
(setState)(C0) 15:15:15 STATE COMM_C -> COMM_C_OK
(setState)(C1) 15:15:15 STATE COMM_C_OK -> C
(I) (setState)(C1) Switching CONTACTOR C2 ON.
(loop)(C1) 0 clients running.
(loop)(C1) 0 clients running.
(loop)(C1) 0 clients running.
(loop)(C1) 0 clients running.
(loop)(C1) 0 clients running.
(operator())(C1) page / (root) requested and sent

Any ideas wjat to look for  ?

dingo35, 2023-09-27:
There's a lot going wrong on your modbus.

Configure Meters only on the Master, and feed only the Master with the API currents.
The Slave will follow the instructions of the Master.

rvdgaag, 2023-09-28:
Disconnected the second EV meter and let SmartEVSE 1 as Master control everything, still nothing works.  Also disconnecting the SMARTEVSE 1 and setting the second unit as solitairy unit did not start charging the car, so I think the unit is not OK.

Here is the log from the MAster + Node1 situation:

Last login: Thu Sep 28 07:58:14 on console
rene@JRFs-MacBook-Air ~ % telnet 192.168.1.48
Trying 192.168.1.48...
telnet: connect to address 192.168.1.48: Connection refused
telnet: Unable to connect to remote host
rene@JRFs-MacBook-Air ~ % 
rene@JRFs-MacBook-Air ~ % telnet 192.168.1.48
Trying 192.168.1.48...
Connected to 192.168.1.48.
Escape character is '^]'.
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: SmartEVSE-6682 IP:192.168.1.48 Mac address:24:4C:AB:3B:19:D4
* Free Heap RAM: 136512
* ESP SDK version: v4.4.3
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter

* Please type the command and press enter to execute.(? or h for this help)
***
v
* Debug: Command received: v
* Debug level set to Verbose
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (6 bytes) 02 04 00 00 00 08 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04
(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Request
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
[... 756 lines cut ...]
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C0) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (6 bytes) 02 04 00 00 00 08 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 02 Function 04
(V) (ModbusDecode)(C0)  Register 0000(ModbusDecode)(C0)  Request
(loop)(C1) 0 clients running.
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A
(I) (printStatus)(C0) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(D) (ModbusDecode)(C0) Received packet(ModbusDecode)(C0)  (23 bytes) 09 10 00 20 00 08 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 09 Function 10
(V) (ModbusDecode)(C0)  Register 0020(ModbusDecode)(C0)  Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A
^C

dingo35, 2023-09-28:
You might want to check your wiring, usually when a unit doesnt start charging its because CP signal to PE is incorrect. 
So check your ground wiring and the CP wiring. 
In State A there should be 12V between CP pin and PE .

rvdgaag, 2023-09-29:
In state A there is 9,6V between CP and PE, also checked it in the working SmartEVSE.

Now I gave up with the open end cable, ordered a wallchassis with a box to connect a factory cable , so to be continued.

dingo35, 2023-09-29:
From tweakers.net, "zelfbouw laadpaal ervaringen":

        As long as no electric car is plugged in, DC + 12 V are simply applied to CP
        When an electric car is plugged in, a diode and a 2.7 kΩ resistor between CP and PE in the type 2 socket of the vehicle pull the + 12 V down to + 9 V.
        The charging station uses this to recognize that a vehicle is connected.
        Now it activates the 1 kHz PWM signal and uses it to signal the car with what maximum charging current it is allowed to charge.

rvdgaag, 2023-10-25:
Issue can be closed now I will open another topic with new findings