serkri/SmartEVSE-3 #209: high WiFi latency

kiss81, 2023-10-23:
I use the api to provide the p1 data to my smartEvse 1.7.3
I send this every 5 seconds. The problem is that it's not always received.
The smartEvse module is literally within 1 meter of the router. I get this ping performance:
round-trip min/avg/max = 6.771/108.583/360.105 ms
I see that sometimes the SmartEVSE get into an error because it didn't receive the p1 data updates in time. This means I can't use the SMART mode now.
Other esp8266 devices have decent ping performance (< 20ms)
Anything I can try?

closed 2023-10-23 by dingo35

dingo35, 2023-10-23:
Just reading that your SmartEVSE is 1 meter from your router: perhaps signal is too strong? 
You could also try feeding through MQTT API.

kiss81, 2023-10-23:
Will try lowering the tx power of the router a bit. Will try the mqtt API as well.