serkri/SmartEVSE-3 #211: Master  / slave issue

rvdgaag, 2023-10-25:
So EVSE1 is master with a mains meter via API, EV meter on modbus, all works fine

EVSE2 is slave 1 with an seperate EV meter on modbus adress 17 (decimal).

EVSE2 does not show in the web-ui the actual charging values of the EV meter, hower when the session ends it report a summary which is correct
[image]

Now the main issue is the EVSE2 doesn't start charging by itself, it only offers OFF and NORMAL in the webui, when you select normal nothing happens. When I switch EVSE 2 with the MQTT interface in HA to SMART it starts charging as it should. EVSE1 shows the right currents, EVSE2 doesn;t display the EV meter.

Attched are log files verbose logging what happens

Master log file

[Master.txt](https://github.com/serkri/SmartEVSE-3/files/13166610/Master.txt)

Slave log file

[slave 1.txt](https://github.com/serkri/SmartEVSE-3/files/13166618/slave.1.txt)

Any ideas?

dingo35, 2023-10-25:
Off of the top of my head, I think the intended behaviour is that all slaves are always in the same mode (Normal/Solar/Smart) as the Master; but Master and/or Slaves can have the Access Bit cleared (mode=OFF) or set (mode=preset mode).

There is one other user who has also master/slave config, he entered a PR that aimed to get rid of illogical/problematical choices on the website; find that PR and get him into this thread, that will speed up things...

rvdgaag, 2023-10-25:
https://github.com/serkri/SmartEVSE-3/pull/152 

Thats him, I will contact him to contribute

jaroslawp, 2023-10-25:
Yes, the intended behaviour seems to be what @dingo35 describes.
If you set Master to SMART this will be propagated to Slave(s): In my setup while I could change via web interface or hardware button the mode of each of my 2 EVSE's separately this always resulted after a while with a modbus 'COMM. ERROR' displayed on both and requiring power cycle. In which mode is your Master when Slave does not start charging automatically ?

rvdgaag, 2023-10-25:
Disable it please. in general I never used it (Normal) to protect my main installation which is limited to 3x25A. The slave should follow the Master. (sounds a bit like S&M to me ;-) 

@jaroslawp  My main was in Smart or off, the slave set to normal did nothing, changing it via HA to Smart made it start.

dingo35, 2023-10-25:
@rvdgaag : so if you would change Master mode from Smart to Normal to Smart back again, the Slave would not end up in Smart mode?

Im not sure if the protocol switches slave modes, I would expect so, but if not, Im not looking forward to changing that protocol since I cant test it.

jaroslawp, 2023-10-25:
For me it seems to work - but generally it is little bit confusing in webui ... slave displays mode 'NORMAL' but still follows the master (starting/stopping charging when I switch between solar/normal/smart). Switching master to off does not change current slave mode when it is already charging.. I guess the slave should just have two states: off and on/follow master ... but not sure it would be any less confusing ...

rvdgaag, 2023-10-25:
@Dingo35 Just tested, the slave follows the master indeed.

But the webui is the confusing factor, off and on/follow master is a good idea.

However when in slave mode the MQTT export should not enable switching the modus.

dingo35, 2023-10-25:
No sorry AFAIK are the modbus routines called irrespective Master or Slave.... 
It will be a search for the variable LoadBalance (0=disabled, 1=master, 2=slave)....

jaroslawp, 2023-10-26:
Hmm.. after further testing I am more convinced that all the mode / current changes on slave EVSE should be removed from webui (and API) ... after few changes between normal/off my slave stops following master and again I have to power cycle it - reboot from webui does not help (some memory/ temp storage is not cleared/reset  ?...) ...

dingo35, 2023-10-26:
It might be connected with @rvdgaag 's EVMeter problen: if the modbus is not read (properly) in the slave, it will no longer follow the master...