serkri/SmartEVSE-3 #216: Lock switch not working

leadenchopsticks, 2023-11-21:
I upgraded from the default EVSE v3 firmware to the latest serkri version. I have a cable connected to the SW/GRD connectors to lock/unlock charging. This does not seem to do anything anymore after the upgrade. Even with the switch set to lock, charging starts when connecting the vehicle. Is this intentional?

dingo35, 2023-11-21:
Could you upload your config (via Raw button on webinterface)?

leadenchopsticks, 2023-11-21:
Sure!

```
{
  "version": "SERKRI-1.7.3",
  "mode": "NORMAL",
  "mode_id": 1,
  "car_connected": false,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "WIFI",
    "rssi": -57,
    "bssid": "12:34:45:67:89:10",
    "auto_connect": false,
    "auto_reconnect": true
  },
  "evse": {
    "temp": 16,
    "temp_max": 65,
    "connected": false,
    "access": true,
    "mode": 0,
    "loadbl": 0,
    "pwm": 1024,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 20,
    "current_main": 25,
    "solar_max_import": 0,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "modem": "Not present",
    "mains_meter": "Disabled",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "",
    "port": 1883,
    "topic_prefix": "SmartEVSE-23429",
    "username": "",
    "password_set": false,
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Disabled",
    "address": 12,
    "import_active_power": 0,
    "total_kwh": 0,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 0,
    "L1": 0,
    "L2": 0,
    "L3": 0,
    "last_data_update": 0,
    "charging_L1": false,
    "charging_L2": false,
    "charging_L3": false,
    "original_data": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  }
}
```

dingo35, 2023-11-21:
What is the configuration of the switch setting (Disable/Access B/Access S/Smart Solar B/Smart Solar S)?

What kind of switch is connected, a toggle switch or a push-to-make-contact switch?

leadenchopsticks, 2023-11-21:
I have set it to Disable when running the original firmware. The switch is a toggle switch. 

Could it be the switch setting was reset after the firmware flash?

dingo35, 2023-11-21:
If you set it to "Disable", it will olignore the switch. Set it to Access S to enable/disable access with a toggle switch. Its in the manual.

leadenchopsticks, 2023-11-22:
Sorry I misread the manual, it was indeed already set to Access S. Stopping/starting via the switch was actually working before the serkri install.  I will check on the display if it's still set to Access S.

leadenchopsticks, 2023-11-23:
It looks like a cycle of the API mode selector from Off to Normal fixed the issue. The switch is now working again.

closed 2023-11-23 by leadenchopsticks