serkri/SmartEVSE-3 #219: Smart charging not working

juhas1982, 2023-12-05:
Hi all,

i installed the newest 1.7.3 firmware to my SmartEVSE controller, i use Sensorbox plus to have Smart charging. With the old original 3.0.0 firmware the smart charging worked just fine, but after this Serkri update my max current does not follow my limitations set to settings.
Is there some setting to be changed after update or what could be wrong? The Screen and also the status page is showing the correct amperes, but is still goes over the my fuse limits.

BR. 
Juha S

dingo35, 2023-12-05:
Please upload your http::/ip/settings, preferably when the max current is exceeded.
If possible add a screenshot of the webserver at approx. the same moment.

juhas1982, 2023-12-07:
Thanks for quick reply!
I may found the main reason why this current limitation is not working correctly. For some reason controller does not know what phase is used to charge the car. I have screenshot of webserver and setting at the same time showing that it is charging at 10 amps on phase 1 but its really charging at all 3 phases. Screenshot of the cars charging display attached showin that its receiving 3x10A.
In the settings site it says that charging L1 true and all other phases are false. And because the current limit overshoots only at phase 2 it does not do anything because it thinks that the car is not charging at phase 2 at all.

I also managed to get the controller to some weird stage when it was thinkin that it is not charging at all, but the car is still receiving 3x10 amps. In this case the controller is not limitting the charge current at all. I attached sceenshot about this too.

BR.
Juha S
[image]
[image]
[image]

dingo35, 2023-12-07:
You are right, SmartEVSE is thinking it is charging at single phase while actually it is charging on 3 phases.

Some EV's are slow to enable their 3 phases (Tesla, Polestar 2); which EV are you charging?

Also, if you can compile yourself, the newest repository/master branch has its phase detection time extended from 22 to 44 seconds, this might solve your problem.

dingo35, 2023-12-07:
When you don't have an EVMeter connected, the phase detection works like this: it measures the currents through the 3 phases, then it starts charging with 6.0 A per phase, and after 22 (new version: 44) seconds it re-measures the currents on the 3 phases, and it determines which phases have an increase of approx. 6A. Now this process can be easily disturbed by other current-users in the house, but also it requires the EV to actually draw that 6A current. Tesla is known to take up to 20 seconds to draw that current, Polestar 2 takes even longer.

[volvoc40.zip](https://github.com/serkri/SmartEVSE-3/files/13597030/volvoc40.zip)

Please flash the attached files, it is the current master with 44 seconds phase detection delay.

kiss81, 2023-12-07:
I do have the exact same problem with my mg4 luxury. The SmartEVSE is thinking it is charging at single phase while it actually is charging with 3 phases. It now ignores phases L2 and L3. I've seen 30A on one of my phases (and the main fuse triggered).
Can we provide some info to fix this?

kiss81, 2023-12-07:
> Did you try the version I provided above?

I just did: it now stays for a moment at 3 phases during the start but then after a while it switches back to 1 phase:

State:
Charging
Charge:
1 x 11.0 A

I limited the current for now to prevent flipping the mains circuit breaker.
Is there something I can check?

dingo35, 2023-12-07:
[volvoc40_debug.zip](https://github.com/serkri/SmartEVSE-3/files/13604701/volvoc40_debug.zip)
Sorry I thought I had enabled it by default ....

kiss81, 2023-12-07:
will flash your build now. Compiling works by the way, but not sure how to enable wifi debugging myself?

dingo35, 2023-12-07:
You didnt run the telnet session in verbose mode, so we can see it detects only 1 phase, but not WHY it is only detecting 1 phase...

For debug mode, look for DBG variable in include/evse.h

kiss81, 2023-12-07:
@dingo35 ok will tun it again with verbose on until it switches to 1 phase.

Some other thing I see:
I think it acts a bit too fast when using API mode. The p1 data through the API takes some time to update. Now sometimes I see it jumping to 3x16.0. Then it sees after 10 seconds it's over the mains limit and it switches to 3x6.0. Then it switches off completely as the p1 data was not yet updated to the 3x6.0A....
In API mode things should be slowed down a bit I think.

dingo35, 2023-12-07:
You are right, I am planning a change that takes into account the frequency of updating; is not a simple change....

kiss81, 2023-12-07:
@dingo35 

`* Please type the command and press enter to execute.(? or h for this help)
***
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=17.0 A Imeasured=15.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.1, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 21.6 A L2: 15.9 A L3: 15.3 A Isum: 52.8 A
/
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=17.0 A Imeasured=15.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=* Debug: Command received: /
1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.1, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 21.6 A L2: 15.9 A L3: 15.3 A Isum: 52.8 A
v(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=17.0 A Imeasured=15.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.1, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=17.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=17.0 A.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C1) L1: 21.6 A L2: 15.9 A L3: 15.3 A Isum: 52.8 A
(loop)(C1) 0 clients running.
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=17.0 A Imeasured=15.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.1, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 21.6 A L2: 15.9 A L3: 15.3 A Isum: 52.8 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=17.0 A Imeasured=15.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=v1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=17.0 A, Idifference=4.1, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=17.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=17.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 15.9 A IsetBalanced: 17.0 A
(I) (printStatus)(C0) L1: 21.6 A L2: 15.9 A L3: 15.3 A Isum: 52.8 A

(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=17.0 A Imeasured=0.8 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1* Debug: Command received: vv
.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=20.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: 6.3 A L2: 0.8 A L3: 0.1 A Isum: 7.2 A
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=20.0 A Imeasured=0.8 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=20.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 20.0 A
(I) (printStatus)(C1) L1: 6.3 A L2: 0.8 A L3: 0.1 A Isum: 7.2 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=20.0 A Imeasured=0.8 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=20.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: 6.3 A L2: 0.8 A L3: 0.1 A Isum: 7.2 A
/
* Debug: Command received: /
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=20.0 A Imeasured=0.8 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=20.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C0) STATE: B1 Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: 6.3 A L2: 0.8 A L3: 0.1 A Isum: 7.2 A
v
[... cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=20.0 A Imeasured=0.8 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=20.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=20.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=20.0 A.
(I) (printStatus)(C0) STATE: C1 Error: 0 StartCurrent: -4 ChargeDelay: 13 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.8 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: 6.4 A L2: 0.8 A L3: 0.2 A Isum: 7.4 A
`

kiss81, 2023-12-07:
@dingo35 thank you for your help: Let me know if there is something I can do / test!

dingo35, 2023-12-07:
These are the currents measured at the start of the session (6.0 A charge current):
(D) (setState)(C0) Trying to detect Charging Phases START Irms[0]=6.4 A.
(D) (setState)(C0) Trying to detect Charging Phases START Irms[1]=0.7 A.
(D) (setState)(C0) Trying to detect Charging Phases START Irms[2]=0.1 A.

After 44 seconds this is measured:
(D) (Timer1S)(C1) Trying to detect Charging Phases END Irms_[0]=6.3 A.
(I) (Timer1S)(C1) Detected Charging Phases: Charging_Prob[0]=0.
(D) (Timer1S)(C1) Trying to detect Charging Phases END Irms_[1]=0.8 A.
(I) (Timer1S)(C1) Detected Charging Phases: Charging_Prob[1]=0.
(D) (Timer1S)(C1) Trying to detect Charging Phases END Irms_[2]=0.2 A.
(I) (Timer1S)(C1) Detected Charging Phases: Charging_Prob[2]=1.
(D) (Timer1S)(C1) Detected Charging Phases: ChargeCurrent=60, Balanced[0]=60, IsetBalanced=60.
(D) (Timer1S)(C1) Suspect I am NOT charging at phase: L1.
(D) (Timer1S)(C1) Suspect I am NOT charging at phase: L2.
(D) (Timer1S)(C1) Suspect I am charging at phase: L3.
(Timer1S)(C1) Charging at 1 phases, L1=0, L2=0, L3=1.

It should detect that it is charging over 0 phases, but as a security it then assumes 1 phase.....
I suspect that your EM_API feed is _much_ too slow, and it might be reinforced by your MG4 that might take a long time before it actually draws power.

You could try to increase PHASE_DETECTION_TIME in line 86 of include/evse.h, but since it is already on a ridicilously long period of 44s I think that is more of a workaround than a solution.

Your EM_API should be fed with fresh data every 5s to have "normal" behaviour, preferably every 1 or 2 seconds.....

dingo35, 2023-12-07:
Also: did you check your wiring, L1 - L3 correctly connected to the CCS plug?

kiss81, 2023-12-08:
@dingo35 Thank you for helping out!
I have the EM_API feed refresh interval at 5s. BUT I see the dsmr plugin of home assistant is updating too slow (30 seconds now). I will fix that.
I will also double check the wiring: I can fix that easily in my EM_API script. I will test again when both are fixed!

juhas1982, 2023-12-08:
Hi,

i installed the new firmware yesterday evening and tested it. For this one charging experiment it seemed to work perfect. I started and stopped the charging multiple times and it always toke all three phases in control.

I am very pleased that i could get help with this issue so quick and also get final resolution to this issue!
Thank you @dingo35 very much!

BR.
Juha

kiss81, 2023-12-08:
I also found that L1 and L3 was swapped hiere. I will test the volvo release again when the car is back :) I have good hopes now. :)

dingo35, 2023-12-08:
The log you posted has a detection time of 22 seconds, so perhaps you should retry the 44s detection time....

dingo35, 2023-12-08:
The web server screen updates every 5 seconds, this does not mean the SmartEVSE polls every 5 seconds. In fact it doesnt poll at all,  it processes both REST API and MQTT API requests on receipt.

kiss81, 2023-12-09:
> The web server screen updates every 5 seconds, this does not mean the SmartEVSE polls every 5 seconds. In fact it doesnt poll at all, it processes both REST API and MQTT API requests on receipt.

Good to know! I will push the MQTT messages @ 1 second rate and see how it works. Will try 44 seconds again on the next test!

kiss81, 2023-12-13:
Tried the volvo non debug build again with 44 seconds and sometimes that still detects 1 or 2 phases. When I set it to 60 it works great.

dingo35, 2023-12-14:
Moved phase_detection_time to 60 in df45228744

closed 2023-12-14 by dingo35