serkri/SmartEVSE-3 #220: POST body not parsed on POST requests

dbursem, 2023-12-10:
Usually, a  POST requests has a POST body containing the data, either `x-www-form-urlencoded` or `json`. But in the SmartEVSE API you need to do a POST request but provide data using GET parameters (in the URL). This is kind of awkward. 

The issue is `request->getParam()` is called in POST requests, without setting the 2nd parameter (post) to true. 

I'd like to fix this. 

Would a PR be accepted if the behavior is not backward-compatible?

dingo35, 2023-12-11:
I was not aware of this "awkwardness", as were the original authors of the API, Im sure.

Since backward incompatibility would hit al lot of users, including the ones using the homeassistant integration, I wouldnt accept such a PR.

Is there  no way compatibility could be warranteed, e.g. autodetecting which of both methods is used?

dbursem, 2023-12-11:
That'll be a couple lines extra code, I'll see if I can get to it