serkri/SmartEVSE-3 #224: SmartEVSE drawing more current then specified mains

Imaginous, 2023-12-16:
I specified 28A as mains max power.
I specified 28A as circuit
And 28A as MAX

I see the total system is using 29.5 on L1.

Why doesn't it throttle down to stay below 28A when using SMART charging?

dingo35, 2023-12-16:
Could you please post:
-your settings (Raw button)
-a _verbose_ telnet log (press v then press enter)

...so I can look into this?

paulw5, 2023-12-24:
I have exactly the same issue. Already have let the network provider repair a fuse two times this year, last friday again.  Never had it before with my SmartevseV1 EV charger solution.
RAW dump, i reflashed it with org smartevse3 Firmware because of the issue. I have a simple P1 addon box to see the current kWh per phase. 
[image]
[image]
[image]
[image]
[image]

{
    "version": "SERKRI-1.8.0",
    "mode": "SMART",
    "mode_id": 3,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "WFIOT",
        "rssi": -85,
        "bssid": "86:2A:A8:4A:CF:8A"
    },
    "evse": {
        "temp": 32,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 1,
        "loadbl": 0,
        "pwm": 340,
        "solar_stop_timer": 0,
        "state": "Connected to EV",
        "state_id": 1,
        "error": "None",
        "error_id": 0,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 200,
        "override_current": 0,
        "current_min": 6,
        "current_max": 20,
        "current_main": 20,
        "current_max_sum_mains": 60,
        "solar_max_import": 16,
        "solar_start_current": 6,
        "solar_stop_time": 0,
        "enable_C2": "Not present",
        "modem": "Experiment",
        "mains_meter": "Sensorbox",
        "starttime": 0,
        "stoptime": 0,
        "repeat": 0,
        "required_evccid": ""
    },
    "ev_state": {
        "initial_soc": -1,
        "remaining_soc": -1,
        "full_soc": -1,
        "energy_capacity": -1,
        "energy_request": -1,
        "computed_soc": -1,
        "evccid": "",
        "time_until_full": -1
    },
    "mqtt": {
        "host": "",
        "port": 1883,
        "topic_prefix": "SmartEVSE-3469",
        "username": "",
        "password_set": 0,
        "status": "Disconnected"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Disabled",
        "address": 12,
        "import_active_power": 0,
        "total_kwh": 0,
        "charged_kwh": 0,
        "currents": {
            "TOTAL": 0,
            "L1": 0,
            "L2": 0,
            "L3": 0
        },
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "mains_meter": {
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "phase_currents": {
        "TOTAL": 27,
        "L1": 5,
        "L2": 6,
        "L3": 16,
        "last_data_update": 1703418219,
        "charging_L1": false,
        "charging_L2": false,
        "charging_L3": false,
        "original_data": {
            "TOTAL": 27,
            "L1": 5,
            "L2": 6,
            "L3": 16
        }
    },
    "backlight": {
        "timer": 0,
        "status": "OFF"
    }
}

dingo35, 2023-12-24:
Sorry guys, but you can keep uploading screenshots all you want, as long as nobody uploads a verbose telnet log there is nothing I can do for you....

paulw5, 2023-12-24:
Sorry didn’t know. Will do this afternoon. No problem

dingo35, 2023-12-24:
In the mean time, you could test this version:
[limit_charging_on_non_charging_phases.zip](https://github.com/serkri/SmartEVSE-3/files/13761817/limit_charging_on_non_charging_phases.zip)

...which has telnet enabled and limits the charging current ALSO on phases you are not charging on.... Again, totally in the blind here.....

paulw5, 2023-12-24:
And just for some extra information. This morning I put the cooker on and saw the car get less power at phase3. And even stopped charging after i turned the oven on. But at the example I uploaded the screenshots of the max current of 20a was not limited.
It seems like that after an certain event the limiter of the current won't work anymore. 

Do I need the latest version you posted to make a Telnet verbose log or can I use v1.8.  Regards Paul

Imaginous, 2023-12-24:
I have it planned, but it's a bit chaotic at work at the moment. Maybe during Christmas.

As developer I completely understand you're in the dark. 
At the moment it seems to work... I have disabled the EV Metering via the API. 
My EV Metering only reports once a minute. I think this might trigger secondary protection. 

But @paulw5 doesn't have this enabled if I look at his screen shots. 

Happy holidays to you all.
@dingo35, we love your work and I don't want to rush you into anything. It's a hobby.
It's me failing to comply ;)

dingo35, 2023-12-24:
@paulw5 Please use the one posted here.

paulw5, 2023-12-24:
[20231224putty.log](https://github.com/serkri/SmartEVSE-3/files/13762118/20231224putty.log)
Is this enough info? 
I started to garden heaters at L3 several times. EVSE lowered the power to My tesla S (info above is my wife M3) all the time.

dingo35, 2023-12-26:
First of all you have "Modem" set to Experiment, it would surprise me if you have a modem attached.

Second, if I understand you correctly, this version gives you no problem. Why would you send me a log of a non-erroneous situation?

paulw5, 2023-12-26:
GoodMorning and no haven't got a modem attached so will change the setting. 
Haven't had any issues yet but also previous versions sometimes exceed the limit but not always. Hope that I can capture some verbose logging if it's happens. 
In march my network operator have to put a new fuse in and also last Friday. And Sunday I saw that the limiter wasn't working.

paulw5, 2024-01-04:
Last week I tested a lot and seems like the limit is working. 

What I remember and never linked it with the issue. I sometimes saw in web interface the statement charging 
2x XX.Y(phase) in stead of the current view of 3x XX.y