serkri/SmartEVSE-3 #225: Number of phases changing when charging stopped and started again

Imaginous, 2023-12-16:
I have a 35A single phase mains.
I supply mains data via API with arguments L1=value, L2=0,L3=0

If I leave out L2 and L3 it will say check data connection. 

After powering up and start charging the first time it will say SMART1F.
If smart charging stops because of no power and restarts again it shows SMART3F.

Imaginous, 2023-12-17:
It seems kind of random or based on the current? 
How does it determine it? 

Many times 6-16A is SMART3F. Amps above this will be shown as SMART1F. 

I opt for a device setting where you simply specify 1/2/3 phase operation.

kiss81, 2023-12-29:
@Imaginous did you try the version 1.80? Think it solves your issue.

Imaginous, 2023-12-29:
I'm using the version of @dingo35, which solves the external P1 step up only occurring when an external package has been received from the API.

I understood this fix is not in V1.8 yet.