serkri/SmartEVSE-3 #226: SMART mode, not charing at max

Xnygra, 2023-12-17:
This issue was described in #130 , but i never came around to follow up on it, but I still have the issue that SMART mode only gives 3x13A when Mains is set to 20A and MAX to 16A. P1 readings is well below that. See attached logg

Do you have any ideas why, or am I doing something wrong. Thank you and thumbs up for the amazing job you guys are doing. 

`{"version":"17:01:48 @Dec 17 2023","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Useme","rssi":-73,"bssid":"E4:5E:1B:7C:79:1A"},"evse":{"temp":28,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":1,"pwm":221,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":130,"override_current":0,"current_min":6,"current_max":16,"current_main":22,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":5,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.86.36","port":1883,"topic_prefix":"SmartEVSE-49475","username":"mqtt","password_set":1,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":10,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":99,"L1":31,"L2":34,"L3":34,"last_data_update":1702831441,"charging_L1":true,"charging_L2":true,"charging_L3":true,"original_data":{"TOTAL":99,"L1":31,"L2":34,"L3":34}},"backlight":{"timer":120,"status":"ON"}}`
[smartEvse-logg.txt](https://github.com/serkri/SmartEVSE-3/files/13696891/smartEvse-logg.txt)

dingo35, 2023-12-18:
I assume you only have one SmartEVSE, however you configured yours as "Master". Disable LoadBalancing on the LCD screen and repost your log again please...

dingo35, 2023-12-18:
Also, since your current is limited to 13A, this is the default if the maximum current of your cable is not recognized.
When the charging is starting (when State A goes to B goes to C) a message is logged with the string "PP Pin" in it (upper and lowercase matching), please make sure that part is included in the log.

Xnygra, 2023-12-18:
Hello Dingo35. Thank you for reply. 
I, for some unknown reason, had LoadBalacing as Master, even though I only have one charger. I did change that, but with same result. See attached Log. Hopefully it can give you some clues. 

the PP pin: logg: (ProximityPin)(C0) PP pin: 1023 (1955 mV)

`{"version":"17:01:48 @Dec 17 2023","mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Useme","rssi":-72,"bssid":"E4:5E:1B:FB:86:5A"},"evse":{"temp":32,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":221,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":130,"override_current":0,"current_min":6,"current_max":16,"current_main":22,"current_max_sum_mains":600,"solar_max_import":0,"solar_start_current":5,"solar_stop_time":10,"enable_C2":"Not present","modem":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.86.36","port":1883,"topic_prefix":"SmartEVSE-49475","username":"mqtt","password_set":1,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":10,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":420,"L1":140,"L2":138,"L3":142,"last_data_update":1702937217,"charging_L1":true,"charging_L2":true,"charging_L3":true,"original_data":{"TOTAL":420,"L1":140,"L2":138,"L3":142}},"backlight":{"timer":120,"status":"ON"}}`
[Smartevse-logg.txt](https://github.com/serkri/SmartEVSE-3/files/13709024/Smartevse-logg.txt)

dingo35, 2023-12-19:
Well your log says it all:
(ProximityPin)(C0) PP pin: 1023 (1955 mV)
(I) (EVSEStates)(C0) Cable limit: 13A  Max: 16A

The SmartEVSE thinks your charging cable can only handle 13A, so it limits the current to that.
I can see in the log that you have configured your SmartEVSE to a "Socket", as opposed to a Fixed Cable.

So questions are:
1) Do you indeed have a socket installed, with which you must use a loose charging cable?
2) If so, is the behaviour the same with a different charging cable?
3) With your loose cable disconnected on both sides, measure the resistance between PP and PE pin, both at the car side of the cable and at the charger side.

At the charger side this table is valid:
Resistor	Conductor size	Maximum Current
100 Ohm	10 mm2	        63 Ampère*
220 Ohm	6 mm2	        32 Ampère
680 Ohm	2.5 mm2	20 Ampère**
1500 Ohm	1.5 mm2	10 Ampère

Xnygra, 2023-12-19:
Thank you very much for finding the root cause. 

Strange however, because I do have a fixed cable, and the Evse has always been set to fixed cable, but who knows, i might have unknowingly change the setting when i have been playing around. 
out of curiosity, where can you read in the log that the setting is set to Socket?

The cable i have is a 2.5mm2 cable, so it should be fine, but i'll have to go through all the connections to see if there might be an issue somewhere which cause the resistance to increase. I also have a brand new cable for the 2nd SmartEvse that i have not yet came around to install, that I can test with.  

I'll let you know later today when I have done some old classic electric fault findings.

dingo35, 2023-12-19:
[code]if (!Config) {                                                          // Configuration (0:Socket / 1:Fixed Cable)
        //socket
        _LOG_A("PP pin: %u (%u mV)\n", sample, voltage);
    } else {
        //fixed cable
        _LOG_A("PP pin: %u (%u mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)\n", sample, voltage);
    }
[/code]

The log message gives an extra warning when the cable is configured to "fixed"; fell into that pitfall once before :-)

dingo35, 2023-12-19:
When you have a fixed cable, the manufacturer Stegen recommends to not connect the PP pin to the SmartEVSE, so you don't need to test your cable resistances; in case of a fixed cable, the charge current limit is supposed to be configured through MaxCurrent, and not through the resistors.

Xnygra, 2023-12-19:
Changed the setting to fixed, and now it working as it should. On the bright side, i gain some deeper knowledge on how the EVSE works. 

Thank you @dingo35 for helping/guide me.

closed 2023-12-19 by Xnygra