serkri/SmartEVSE-3 #230: MQTT error when starting up

djoenez, 2024-01-03:
Hello,

I found this in my HA log: 
2024-01-03 19:43:50.554 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'EVSE1/MainsCurrentL1': 'verride1'

I think it should mean override?

dingo35, 2024-01-03:
Are you sure your wifi connection is not crippled?

djoenez, 2024-02-01:
Just a quick update. I fixed the wifi issue (both the EVSEs have now a wired AP 1m away from them instead of 8m). and i redid the modbus wiring to make sure everything is ok.

Unfortunately i still get either:
[image]
(you can see in the second block of bouncing values there was actually some charging, so it doesnt bounce to zero but to the Wattage it is charging at (solar mode)

or:
[image]

It doesn't matter if i request this data via REST or MQTT, they both see this kind of behaviour, on both the EVSEs.

Running the last update you send me in our previous conversation. I'll update to 3.5 later today to see if the issue persists.