serkri/SmartEVSE-3 #231: current_max_sum_mains does not work on my install

bartv, 2024-01-07:
I live in Belgium and have been trying the current_max_sum_mains setting but there are some issues with it. My charger has 3 phases but I actually only use 1 phase because it allows me to use lower charge current in solar mode. The setting was always off but it would never  start unless I set it to "not installed". The minimal charging current is 8A because my Ioniq 5 trips the RCB when it modulates down to 6A

The second issue is that if it starts it only works for a few minutes until it goes into a "No Power Available" state that I can only fix by upping the sum config option to a high value and rebooting the evse. I installed the debug version and collected some logs.

My raw data
```json
{
    "version": "SERKRI-1.8.0",
    "mode": "SMART",
    "mode_id": 3,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "building",
        "rssi": -59,
        "bssid": "82:8A:20:5A:BB:D0"
    },
    "evse": {
        "temp": 29,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 1,
        "loadbl": 0,
        "pwm": 1024,
        "solar_stop_timer": 0,
        "state": "Charging Stopped",
        "state_id": 9,
        "error": "No Power Available",
        "error_id": 1,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 320,
        "override_current": 0,
        "current_min": 8,
        "current_max": 32,
        "current_main": 32,
        "current_max_sum_mains": 28,
        "solar_max_import": 0,
        "solar_start_current": 0,
        "solar_stop_time": 2,
        "enable_C2": "Not present",
        "modem": "Not present",
        "mains_meter": "API",
        "starttime": 0,
        "stoptime": 0,
        "repeat": 0
    },
    "mqtt": {
        "host": "192.168.33.2",
        "port": 1883,
        "topic_prefix": "smartevse-2",
        "username": "",
        "password_set": 0,
        "status": "Connected"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Eastron3P",
        "address": 12,
        "import_active_power": 0,
        "total_kwh": 2379.7,
        "charged_kwh": 0,
        "currents": {
            "TOTAL": 0,
            "L1": 0,
[... 554 lines cut ...]
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(loop)(C1) 0 clients running.
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A
(I) (printStatus)(C1) L1: 0.9 A L2: 5.9 A L3: 3.5 A Isum: 10.3 A
(I) (printStatus)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 5.9 A IsetBalanced: 10.0 A

```

bartv, 2024-01-07:
When I raise the sum to a higher value like 40A both smart and normal mode work fine but that is too high for me

```json
{
"version": "SERKRI-1.8.0",
"mode": "NORMAL",
"mode_id": 1,
"car_connected": true,
"wifi": {
"status": "WL_CONNECTED",
"ssid": "building",
"rssi": -62,
"bssid": "82:8A:20:5A:BB:D0"
},
"evse": {
"temp": 30,
"temp_max": 65,
"connected": true,
"access": true,
"mode": 0,
"loadbl": 0,
"pwm": 545,
"solar_stop_timer": 0,
"state": "Charging",
"state_id": 2,
"error": "None",
"error_id": 0,
"rfid": "Not Installed"
},
"settings": {
"charge_current": 320,
"override_current": 0,
"current_min": 8,
"current_max": 32,
"current_main": 32,
"current_max_sum_mains": 40,
"solar_max_import": 0,
"solar_start_current": 0,
"solar_stop_time": 2,
"enable_C2": "Not present",
"modem": "Not present",
"mains_meter": "API",
"starttime": 0,
"stoptime": 0,
"repeat": 0
},
"mqtt": {
"host": "192.168.33.2",
"port": 1883,
"topic_prefix": "smartevse-2",
"username": "",
"password_set": 0,
"status": "Connected"
},
"home_battery": {
"current": -107,
"last_update": 1704658375
},
"ev_meter": {
"description": "Eastron3P",
"address": 12,
"import_active_power": 7.3,
"total_kwh": 2379.9,
"charged_kwh": 0.1,
"currents": {
"TOTAL": 312,
"L1": 0,
"L2": 0,
"L3": 312
},
"import_active_energy": 2379.9,
"export_active_energy": 0
},
"mains_meter": {
"import_active_energy": 0,
"export_active_energy": 0
},
"phase_currents": {
"TOTAL": 313,
"L1": -31,
"L2": 35,
"L3": 309,
"last_data_update": 1704658372,
"charging_L1": false,
"charging_L2": false,
"charging_L3": false,
"original_data": {
"TOTAL": 313,
"L1": -31,
"L2": 35,
"L3": 309
}
},
"backlight": {
"timer": 120,
"status": "ON"
}
}
```

dingo35, 2024-01-08:
First of all, just to make sure, you DO have a seconds contactor installed and wired according to this schematics: [https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/installation.md](url)

Second, you have your Solar mode misconfigured: your StartCurrent is set at 0A, which means that Charging will start as soon as your house is delivering current to the grid.
Then you have set your MinCurrent to 8A, so you will need 8A to start charging. But then you have Solar_Max_Import set at 0A, so you are prohibiting your SmartEVSE to import any current from the net. 

So you are basically telling your SmartEVSE with this config to start charging as soon as you are delivering 8A to the grid (which with current solar conditions seem to be hard to reach), and if the delivered current drops stop charging within 2 minutes.

Mind you, I'm not saying there is not a bug in there, I'm just saying I can trace anything useful with this kind of config.

I suggest:
1) you set StartCurrent and Solar_Max_Import at logical values, AT LEAST MinCurrent A apart from eachother.
2) I am currently working on the new version 3.5 in a new repo, please flash this newer version attached here and report 
 back in a new issue in [https://github.com/dingo35/SmartEVSE-3.5](url)
[firmware3.5-5769035d5adb.zip](https://github.com/serkri/SmartEVSE-3/files/13857812/firmware3.5-5769035d5adb.zip)

bartv, 2024-01-08:
> First of all, just to make sure, you DO have a second contactor installed and wired according to this schematics: [https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/installation.md](url)?

I wired it indeed up like the documentation and it works all fine when using 3 phase mode. The problem I noticed is that it seems to use 3x the base current even if C2 is set to always off instead of 1x the base current. 

> Second, you have your Solar mode misconfigured: your StartCurrent is set at 0A, which means that Charging will start as soon as your house is delivering current to the grid. Then you have set your [...]

I will try to have a look this evening.

dingo35, 2024-01-08:
I rewrote the IsCurrentAvailable routine, that was not aware of the EnableC2 setting. I cannot test it myself currently, but this version should behave better when in single phased solar mode:
[firmware3.5-d76d9fa51f.zip](https://github.com/serkri/SmartEVSE-3/files/13861926/firmware3.5-d76d9fa51f.zip)

bartv, 2024-01-08:
I tried this new version and both issues seem to be fixed. It now starts when there is a 8A available and not when there is 3x 8A available.

I changed some of the solar settings (which is only possible when in solar mode). I have set the max import to 16A and when using smart mode I could see it charging at 20A. I have let it run for 5 minutes and it nicely regulated the charge current up and down.

This evening when off-peak hours start I will do a longer session. Do you want me to collect logs over telnet? And if yes, at which log level? Because at debug mode there are a lot modbus messages.

dingo35, 2024-01-08:
I only need logs when you run into unexpected behaviour; the preferred level is verbose, which is currently the default.

Thx for testing!

bartv, 2024-01-08:
I noticed another potential issue. All my breakers are 32A (main + ev). I configured the sum of all phases to 30A. When I start charging there is already 1.3A consumption on L3 (which is L1 for the car)

[image]

When I then start it in smart mode, it starts at 32A overshooting the limits for a very brief moment:

[image]

and then it quickly regulates down:

[image]

It is not a big problem for the capacity rate but it could potentially trip a fuse if there was already a high current on that phase.

dingo35, 2024-01-08:
I would appreciate a verbose debug log from the start of the charging process until it stabilizes at 32A...

bartv, 2024-01-10:
Here are the logs of a charging session I just started. You can clearly see the overshoot here by starting at 32A:

```
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.3 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.3 A L3: 3.2 A Isum: 11.2 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(operator())(C1) page / (root) requested and sent
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(I) (printStatus)(C1) STATE: B1 Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(ProximityPin)(C0) PP pin: 1023 (1864 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
(I) (EVSEStates)(C1) Cable limit: 32A  Max: 32A
(setState)(C1) 21:07:28 STATE B1 -> B
(EVSEStates)(C0) Diode OK
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 33.5 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(setState)(C1) 21:07:32 STATE B -> C
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 32.0 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.2 A Isum: 11.3 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 7.4 A IsetBalanced: 32.0 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.4 A L3: 3.3 A Isum: 11.4 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 34.3 A IsetBalanced: 27.8 A
(I) (printStatus)(C1) L1: 0.9 A L2: 7.4 A L3: 34.3 A Isum: 42.6 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 30.0 A IsetBalanced: 25.2 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.3 A L3: 30.0 A Isum: 38.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 27.5 A IsetBalanced: 23.4 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.3 A L3: 27.5 A Isum: 35.5 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 25.6 A IsetBalanced: 22.2 A
(I) (printStatus)(C1) L1: 0.8 A L2: 7.2 A L3: 25.6 A Isum: 33.6 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 24.6 A IsetBalanced: 21.3 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.5 A L3: 24.6 A Isum: 32.8 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 23.6 A IsetBalanced: 20.7 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.5 A L3: 23.6 A Isum: 31.8 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 23.7 A IsetBalanced: 20.1 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.5 A L3: 23.7 A Isum: 31.9 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 23.1 A IsetBalanced: 19.8 A
(I) (printStatus)(C1) L1: 0.7 A L2: 7.2 A L3: 23.1 A Isum: 31.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.5 A IsetBalanced: 19.6 A
(I) (printStatus)(C1) L1: 0.9 A L2: 7.2 A L3: 22.5 A Isum: 30.6 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 22.1 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 22.1 A Isum: 30.3 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 21.9 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 21.9 A Isum: 30.1 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 21.8 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 21.8 A Isum: 30.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 21.8 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 21.8 A Isum: 30.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 21.8 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 21.8 A Isum: 30.0 A
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -8 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 21.8 A IsetBalanced: 19.5 A
(I) (printStatus)(C1) L1: 1.0 A L2: 7.2 A L3: 21.8 A Isum: 30.0 A
```

Here are my current settings:
[settings.json](https://github.com/serkri/SmartEVSE-3/files/13893292/settings.json)

dingo35, 2024-01-10:
Excellent log, except, it is not "verbose"; in your telnet session, press v then enter; then we can see "checkpoint" messages that clarify at which point in the code the overshoot is made....

Appreciate your testing!!

bartv, 2024-01-10:
Verbose is of course not the same as info :man_facepalming: 

[verbose.log](https://github.com/serkri/SmartEVSE-3/files/13893795/verbose.log)

Glad I can help. I really need this feature so your work is much appreciated.

bartv, 2024-01-10:
Once it regulates down to the target 30A it follows it quite well. Whenever my heatpump de-ices the current goes up and then when it starts again the load current goes down nicely.

dingo35, 2024-03-01:
Ok so your second issue is solved now?

"The second issue is that if it starts it only works for a few minutes until it goes into a "No Power Available" state that I can only fix by upping the sum config option to a high value and rebooting the evse. I might be missing something in the settings but I cannot figure out what. I installed the debug version and collected some logs."

And you are only worried about the regulation taking a few seconds, right?

bartv, 2024-03-04:
I think this is fixed now. I am currently using your 3.5.0-rc0 fork. I am seeing some other changes related to battery. I will report them on that repo.

closed 2024-03-04 by bartv