serkri/SmartEVSE-3 #232: Ford MachE charging issue

Mawverick, 2024-01-10:
SmartEVSE-3-1.5.3 firmware, 1-phase contactor.

SmartEVSE wont charge Ford MachE. The SmartEVSE is charging the eTransit and Ionic 5. But wont charge a MachE. Is there any reason this could be? It happened at two different locations now.

dingo35, 2024-01-10:
I see no reason for that; I recommend you 
-upgrade to version v1.8.0, debug version
-post your http://ipadress/settings
-post the telnet debug log of the start of your session, then we can see whats going on...

dingo35, 2024-01-10:
Since you are single phase charging, you might want to check whether your single phase is connected to the L1 pin of your connector. Some EV's are picky about that.

Mawverick, 2024-02-26:
This is the data from the charger when i plugged in (2x).  single phase is connected to the L1 pin of the Type 2 connector. 

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.

(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.

(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(loop)(C1) 0 clients running.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.

(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.

(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

[... 986 lines cut ...]
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0.

(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=32.0 A.

(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=32.0 A.

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 32.0 A

(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=32.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=0

dingo35, 2024-02-26:
Your EV is never detected as "Connected" by the EVSE; this might be a wiring problem, the CP pin is not connected (correctly) to your EVSE or the plug is not put in far enough in your E-Mach.

Explanation of the CP signal you can find here:
[https://gathering.tweakers.net/forum/list_messages/1648387](url)

You could upgrade to current firmware version 3.5.0-RC1 but I don't think that will solve your problem....

Mawverick, 2024-02-27:
Once the connection with the charger established the car says "Charging station paused" 
[image]
Must check whether we are not experience an issue described below
https://www.macheforum.com/site/threads/charging-paused.6827/

dingo35, 2024-02-27:
That message also refers to "pushing the plug in one more millimeter". Did you try pushing the plug in your Ford EV as hard as you could?

Mawverick, 2024-02-27:
Not that much. 
https://github.com/serkri/SmartEVSE-3/assets/150398493/b5c30457-8d1d-459b-b5c7-9a9c7c2416f3
Will check it out if possible to push the plug in a little further

Mawverick, 2024-04-03:
We have plugged in and out of the car over several times and it hasn't charged once. I even swapped the plug and cable from a charger that was working with the car to the charger and it didn't work.
I have tested the CP from the plug to the charger and its all good.
I removed the RCD and It didn't change anything.
I also checked that all the pins were going in and that the plug was fully pushed in.

When I had the car plugged in and tested CP - Earth, when I removed my test leads the charger started charging. I did this 3-4 times and the same thing happened. 

https://github.com/serkri/SmartEVSE-3/assets/150398493/f192a4b2-491f-4049-b4b8-095a6ec0dcf6

See video attached.