serkri/SmartEVSE-3 #233: platformio run failed

Zakarumit, 2024-01-12:
Hi, 
I have tried compiling the latest version but looks like there is some kind of problem. I enclose the log - I will be grateful for any suggestion or solution.

`$ platformio run
Processing release (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/maxgerhardt/ESPAsyncDNSServer#patch-1
git version 2.43.0.windows.1
Cloning into 'C:\Users\zakar\.platformio\.cache\tmp\pkg-installing-yoo54zm4'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 13 (delta 1), reused 4 (delta 0), pack-reused 0
Receiving objects: 100% (13/13), 4.68 KiB | 4.68 MiB/s, done.
Resolving deltas: 100% (1/1), done.
Library Manager: ESPAsyncDNSServer@1.0.1+sha.3ccb2d3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ ^3.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing esphome/AsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.0 has been installed!
Library Manager: Installing khoih-prog/ESPAsync_WiFiManager @ 1.14.1
Unpacking  [####################################]  100%
Library Manager: ESPAsync_WiFiManager@1.14.1 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing me-no-dev/AsyncTCP @ >=1.1.1
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@1.1.1 has been installed!
Library Manager: Installing me-no-dev/ESP Async WebServer @ >=1.2.3
Unpacking  [####################################]  100%
Library Manager: ESP Async WebServer@1.2.3 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing khoih-prog/ESP_DoubleResetDetector @ >=1.3.2
Unpacking  [####################################]  100%
Library Manager: ESP_DoubleResetDetector@1.3.2 has been installed!
Library Manager: Installing miq19/eModbus @ 1.7.0
Unpacking  [####################################]  100%
Library Manager: eModbus@1.7.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing git+https://github.com/maxgerhardt/Ethernet.git
git version 2.43.0.windows.1
Cloning into 'C:\Users\zakar\.platformio\.cache\tmp\pkg-installing-ahem7rj7'...
remote: Enumerating objects: 47, done.
remote: Counting objects: 100% (47/47), done.
remote: Compressing objects: 100% (40/40), done.
Receiving objects: 100% (47/47), 52.69 KiB | 999.00 KiB/s, done.

Resolving deltas: 100% (4/4), done.
Library Manager: custom-Ethernet@2.0.0+sha.c9399a9 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ ^6.21.4
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.21.5 has been installed!
Library Manager: Installing git+https://github.com/256dpi/arduino-mqtt#7afcfb1
git version 2.43.0.windows.1
Cloning into 'C:\Users\zakar\.platformio\.cache\tmp\pkg-installing-lr74de33'...
remote: Enumerating objects: 2831, done.
remote: Counting objects: 100% (145/145), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 2831 (delta 73), reused 119 (delta 57), pack-reused 2686Receiving objects: 100% (2831/2831)
Receiving objects: 100% (2831/2831), 475.35 KiB | 1.92 MiB/s, done.
Resolving deltas: 100% (1733/1733), done.
HEAD is now at 7afcfb1 added methods to support QoS1/2 retry logic
Library Manager: MQTT@2.5.1+sha.7afcfb1 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.6+sha.b63f947
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncDNSServer @ 1.0.1+sha.3ccb2d3
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPAsync_WiFiManager @ 1.14.1
|-- eModbus @ 1.7.0
|-- ArduinoJson @ 6.21.5
|-- MQTT @ 2.5.1+sha.7afcfb1
|-- Preferences @ 2.0.0
|-- AsyncTCP-esphome @ 2.1.0
|-- ESP Async WebServer @ 1.2.3
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- Update @ 2.0.0
|-- WiFi @ 2.0.0
|-- RemoteDebug @ 3.0.5
Building in release mode
Compiling .pio\build\release\src\OneWire.cpp.o
Compiling .pio\build\release\src\evse.cpp.o
Compiling .pio\build\release\src\font.cpp.o
Compiling .pio\build\release\src\font2.cpp.o
Compiling .pio\build\release\src\glcd.cpp.o
Compiling .pio\build\release\src\modbus.cpp.o
Compiling .pio\build\release\src\utils.cpp.o
Building .pio\build\release\bootloader.bin
Generating partitions .pio\build\release\partitions.bin
esptool.py v4.4
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\release\lib98a\AsyncUDP\AsyncUDP.cpp.o
Compiling .pio\build\release\libdfd\ESPAsyncDNSServer\ESPAsyncDNSServer.cpp.o
Compiling .pio\build\release\lib5d5\AsyncTCP-esphome\AsyncTCP.cpp.o
Compiling .pio\build\release\libfe4\FS\FS.cpp.o
Compiling .pio\build\release\libfe4\FS\vfs_api.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFi.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFiAP.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFiClient.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFiMulti.cpp.o
Archiving .pio\build\release\libdfd\libESPAsyncDNSServer.a
.pio/libdeps/release/AsyncTCP-esphome/src/AsyncTCP.cpp:753:2: error: #else without #if
 #else
  ^~~~
.pio/libdeps/release/AsyncTCP-esphome/src/AsyncTCP.cpp:755:2: error: #endif without #if
 #endif
  ^~~~~
.pio/libdeps/release/AsyncTCP-esphome/src/AsyncTCP.cpp:1020: error: unterminated #if
 #if LWIP_IPV4 && LWIP_IPV6

Compiling .pio\build\release\lib1d2\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\release\lib1d2\WiFi\WiFiScan.cpp.o
Archiving .pio\build\release\lib98a\libAsyncUDP.a
Compiling .pio\build\release\lib1d2\WiFi\WiFiServer.cpp.o
.pio/libdeps/release/AsyncTCP-esphome/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio/libdeps/release/AsyncTCP-esphome/src/AsyncTCP.cpp:754:39: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
         return connect(IPAddress(addr.addr), port);
                                       ^~~~
                                       u_addr
*** [.pio\build\release\lib5d5\AsyncTCP-esphome\AsyncTCP.cpp.o] Error 1
src/evse.cpp: In function 'void loop()':
src/evse.cpp:4664:50: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
         _LOG_A("Time not synced with NTP yet.\n");
                                                  ^
============================================= [FAILED] Took 16.74 seconds =============================================`

PhilippeRigaux, 2024-01-28:
I ran into a lot of errors to get the build running flawlessly with the lastest version of platformio. So I installed an older version that work pretty well..

`cd $HOME
rm -rf .platformio/
pip3 install -U https://github.com/platformio/platformio-core/archive/refs/tags/v6.1.5.zip
git clone https://github.com/serkri/SmartEVSE-3.git
cd SmartEVSE-3/SmartEVSE-3/
platformio run
cd .pio/build/release`

Zakarumit, 2024-02-10:
> I ran into a lot of errors to get the build running flawlessly with the lastest version of platformio. So I installed an older version that work pretty well..
>
> `cd $HOME` `rm -rf .platformio/` `pip3 [...]

Thank you for the help. I tried it but I am stuck: 
$ platformio run
bash: platformio: command not found

...I am on win10 and entering the command from the Git Bash, i.e. MINGW64. Looks like the installation went OK and I think its located in 
C:\Users\zakar\AppData\Roaming\Python\Python311\site-packages\platformio
I can also see the platformio.exe in
C:\Users\zakar\AppData\Roaming\Python\Python311\Scripts

Could you help please?