serkri/SmartEVSE-3 #236: SMART mode stops charging

kiss81, 2024-01-20:
When I charge in SMART mode and the total current on one phase goes (a little bit) over the limit I set. (25A) the charging current goes down as expected, which is great as it prevents the mains fuses to blow. :) 
It starts going down from 16A to i.e. 11A and then 6A. Unfortunately it also ends up stopping the charge for a minute and then restart again. I do think the smartEVSE is overreacting a bit. I would prefer that it always goes to i.e. 6A mode and slowly increase from there if there is room for it. If 6A is still too much it should shut off charging of course.

Not sure if stopping the charge is an issue or if the algorythm reacts too fast. If it's the latter it would be nice to have a configuration setting in the menu for the algo: fast, normal, slow or an algo loop speed or something.

I update my P1 data using mqtt every 1 second so that should deliver the data fast enough?

dingo35, 2024-01-20:
When less then MinCurrent is available, the SmartEVSE has no other choice then to stop charging...

dingo35, 2024-01-20:
You could try one of the 3.5 alpha releases floating around, it has several improvements....

dingo35, 2024-01-20:
One of the improvements is that it will only update the charge current if new data comes in.

Mind you, you should only upload "fresh" data through your MQTT interface; so if your P1 interface gives new data every 5s, only upload every 5s !!

kiss81, 2024-01-20:
Thanks for all the info: I will test and report back!

negronoir, 2024-01-24:
> Thanks for all the info: I will test and report back!
> 
> @dingo35 No luck with Smart EVSE V3 Version: 15:31:52 @jan 8 2024 The display tells me access denied and I cannot control the charge mode. Can I build the latest myself maybe? (which branch)
> 
> edit 2: Moving back to a slower interval (10 seconds) seems to work quite well! I found that turning on the dishwasher can create a peak of 19A on one of the phases. As the minimal charge current is 6A (can't go lower?) there is no room for charging anymore. Therefore I now set my mains to 26A (small risk the fuse will blow one time... :P)

For the edit2

As @dingo35 said, stop charging under 6A is normal behavior.
Standard min charge is 6A but some cars (e.g. Tesla) allow 5A charge
I've rebuild v1.8.3 to allow min charging at 5A
If you need the source code changes, I'll provide you

dingo35, 2024-01-24:
There is no version 1.8.3

If your LCD screen says "access denied" you need to press the menu button or choose the mode on the webserver. Please RTFM : 
https://github.com/dingo35/SmartEVSE-3.5

kiss81, 2024-01-29:
This was minly a user / configuration error so I will close this one

closed 2024-01-29 by kiss81

negronoir, 2024-01-30:
> There is no version 1.8.3
> 
> If your LCD screen says "access denied" you need to press the menu button or choose the mode on the webserver. Please RTFM : https://github.com/dingo35/SmartEVSE-3.5

Sorry! 
I have made a mistake. I wanted to say both v1.7.3 & v1.8.0