serkri/SmartEVSE-3 #244: WiFi settings not stored after reboot

odem881, 2024-02-19:
Every time i reboot/restart SmartEVSE it does not connect to the WiFi so i need to perform the procedure of starting AP and inputting WiFi details to it.

Why it doesn't store the connection details?

dingo35, 2024-02-19:
It does store the credentials.

Just to make sure, upgrade to latest version https://github.com/dingo35/SmartEVSE-3.5/releases/tag/v3.5.0-RC0

Goto Menu/Wifi/Setup and follow the _exact_ instructions . 

If your other settings arent stored either you might have a hardware problem...

dingo35, 2024-02-19:
You sure you set the Wifi option to Enabled?

odem881, 2024-02-19:
That is strange, i kept it running for a while and perform the reboot again and now it works. Not sure what happened. Thank you anyway.