serkri/SmartEVSE-3 #247: EV Meter charged energy glitch

camel1cz, 2024-02-28:
After some time, I finally got my custom EV meter working.
I have only EV Meter, no Mains Meter, no API... using Normal charging mode. Using 1 phase changing.

It acts like this:
- firmware can correctly read energy from EV Meter when not charging (doesn't matter if car is connected or not),
- when charging, the total energy gets decreased by constant (it's remarkably similar to the charging current),
- each several seconds, the total energy gets decreased by a huge number - in my case close to 3600 (in my case the total gets negative) - in the same time the currents measured are nonsense negative values too,
- after the charging ends, everything gets fixed, the EV meter shows correct energy updated with last charging,
- when not charging, the total energy is stable, nothing changes and is consistent with EV meter LCD.

Can anyone help me, where to look for a problem? I'm able to compile the firmware and can understand some C/C++

From the telnet debug, I can see just following strange messages written in orange (not sure if related to the problem) - they are take while not charging:

```
(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 9.0 A
(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(I) (printStatus)(C1) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 9.0 A
(I) (printStatus)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A

```

Any help is appreciated!

dingo35, 2024-02-28:
Upgrade to 3.5.0-RC1, I think the problem is less in that version, but not completely solved. We're working at it....

camel1cz, 2024-02-29:
Thanks for you reply and your good work in this project, @dingo35 !

I'm aware of your fork and like it, but am a bit worried about compatibility - will try it out today, hopefully.

Btw. would you mind to describe in your documentation main differences between stock, serki and your version? And ideally also update tutorial (is config compatible? How to purge it if needed)
I noticed at least a drop of modbus TCP proxy, which I don't use... but differences seem to be huge (200+ commits ahead atm) which could be scary for someone using wallbox in production.

Btw. I'm aware of at least one problem in the code and would be happy do create PR.

camel1cz, 2024-02-29:
OK, I tried out your version and it behaves very similar to serkri version.

New is the fact, your version doesn't let me configure EV Meter while not having mode other than NORMAL which requires Mains Meter set.

For now I didn following changes to your version:
- fixed Custom modbus setup which didn't work for me,
- fixed processing of modbus messages when not having Mains Meter
- changed menu to always show Mains and EV meter config (rough fix to get it usable for my scenario)

Now I would need to block all energy calculations conditionally when MainsMeter is disabled - but I'm not sure if it suits your plans?

dingo35, 2024-03-01:
Your use-case is very specific; usually users have either NO kWh Meter attached, and they are just simply charging in Normal Mode. OR they have one attached, as MainsMeter, because they need to guard their main breaker. And sometimes, as a "luxury", they add an EVMeter to keep track of charged kWh .

Want you want to achieve is against the whole architecture of the code, so I would discourage you to put any effort in this; the only thing you will achieve is showing your EVMeter values on the webserver of the SmartEVSE; it will not do anything with it.....

So rewire your EVMeter to a MainsMeter, then it serves a greater purpose for your SmartEVSE, or walk over to your current EVMeter to read its values :-)

camel1cz, 2024-03-01:
Thanks for your hint - it makes sense, indeed!

What will happen if I configure my energy meter as mains meter and set EV meter to disabled? Will I see the kWh charged in the HA - that's currently I only concern.

camel1cz, 2024-03-01:
OK, the simplest solution to get energy charged into HA seems to be either standalone ESP in ESPHome or custom patch for smartevse firmware... I'll probably do the patch and keep it outside the development tree.

Thank you very much for your consultation!

dingo35, 2024-03-01:
If that is your goal, buy a simple Rs485 to USB adapter to connect your EVmeter to your HA server. Google for modbus + homeassistant.

camel1cz, 2024-03-01:
Sure it's a solution, but for me it's much easier to change 3 lines in smartevse firmware and pass the kWh unchanged.

Btw. I'm not getting why is the amount of total energy calculated from imported - exported... btw. the error I found comes from exactly this calculation. The imported energy is correct but the exported gets from time to time equal to the instant power (in Watts).

dingo35, 2024-03-01:
Then I suspect your meter config is erroneous.

camel1cz, 2024-03-01:
Could be as the custom setup of energy meter doesn't allow me to set exported energy modbus register - so it could read it by "itself" from "somewhere" getting the value for power and deducting it from the imported energy.

Anyway the custom configuration is totally wrong and it sets all registers to zero if you set any of the registers to 0x08 (for me it's reading for current) - this code is titled "Backward compatibility < 2.20" in evse.cpp on line 3415