serkri/SmartEVSE-3 #248: Feature Request: access per API

systeembeheerder, 2024-03-01:
Could the access setting be made available via the API?

Usercase: Connected is a pushbutton switch inside my home to enable/disable access to the charging station. I would like to set disable access as soon as the EVPlugState changes to Disconnected. So if someone unplugs the EV access should be denied until i push the button again.

systeembeheerder, 2024-03-17:
til changing the charging mode also changes the access mode. :(

which is kind of odd because there are both separate defined in the API

dingo35, 2024-03-17:
If you set mode to 0 through the API, access is denied. If you set it 1 - 3 the different modes are switched.

You also have charging state and if the vehicle is connected, so you can script it all, RTFM:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/REST_API.md

systeembeheerder, 2024-03-17:
you can RTFM all you want; but in [the manual](https://github.com/SmartEVSE/SmartEVSE-3/blob/main/manual/SmartEVSEv3_install_v3.0.pdf) does not suggest there is any relation to access and mode.  

Even [your own API](https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/REST_API.md) gives different fields for mode and acces:

```
  "version": "21:02:46 @Jan  3 2024",
  "mode": "OFF",
  **"mode_id": 0,**
  "car_connected": false,
  "evse": {
    "temp": 16,
    "temp_max": 60,
    "connected": false,
    **"access": false,**
    "mode": 1,   
    "loadbl": 0,
    "pwm": 1024,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
...
```

Based on that; asking to be able to change those settings independent for each other isn't that weird is it? 
My hardware switch (see manual) changed access, not mode.