serkri/SmartEVSE-3 #251: Feature Request: Loadbalancing use-case

vdweegen, 2024-03-22:
Hi all,

First of all, I've been using the SmartEVSE with Serkri for a while now, and I've been extremely happy. Fantastic work!

I'm looking for some help w.r.t. a specific load-balancing use-case, as I've recently decided to upgrade my setup by replacing my old 1-Phase "dumb EVSE" with another SmartEVSE unit.

So, currently i've got two SmartEVSE's, both running Serkri 1.8.0.

Both SmartEVSE's are connected via modbus, to each other, to a Sensorbox, and to two kWh meters (Eastron SDM72D-m).

LoadBalancing is configured, SmartEVSE 1 is Master, SmartEVSE 2 is configured as Node-1 (Slave).

SmartEVSE 1 has a fixed 16A, 3-Phase cable which is controlled by Relay 1, SmartEVSE 2 has a fixed 16A, 1-Phase cable which is controlled by Relay 2 (L1).

While loadbalancing is working correctly by making sure the ChargeCurrent on both units together is not greater than CIRCUIT, I was hoping if would be possible to have the Slave Unit have its own kWh-meter configured. I've noticed that when SmartEVSE2 is configured as Node1 (Slave), it is not actually using the configured kWh-meter (The configured kWh-meter values are also not used to display or even updated in the ev-meter->current fields for the slave-unit).

My setup is as follows.

                       mains
                         |
                  [main breaker 3x25A]
                         |
                  [kWh meter SENSORBOX]
                         |
      -----------------------------------
      |            |                    |
                              [smartevse breaker 3x16A]
                                        |
                                  ----------------
                                  |              |
                            [kWh meter 1]  [kWh meter 2]
                                  |              |
                            [relay 1]         [relay 2]
                                  |              |
                             [car 1]           [car 2]

Configurations are:
// SmartEVSE1
`
{
  "version": "SERKRI-1.8.0",
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": false,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "-",
    "rssi": -62,
    "bssid": "-"
  },
  "evse": {
    "temp": 25,
    "temp_max": 65,
    "connected": false,
    "access": true,
    "mode": 1,
    "loadbl": 1,
    "pwm": 1024,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 86,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_sum_mains": 600,
    "solar_max_import": 0,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "modem": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "xxx.xxx.xxx.xxx",
    "port": 1883,
    "topic_prefix": "SmartEVSE-xxxxx",
    "username": "xxxxxx",
    "password_set": 0,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 101,
    "import_active_power": 0,
    "total_kwh": 891.2,
    "charged_kwh": 14.8,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 891.2,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 159,
    "L1": 140,
    "L2": 6,
    "L3": 13,
    "last_data_update": 1711131319,
    "charging_L1": false,
    "charging_L2": false,
    "charging_L3": false,
    "original_data": {
      "TOTAL": 159,
      "L1": 140,
      "L2": 6,
      "L3": 13
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  }
}
`
// SmartEVSE2
`
{
  "version": "SERKRI-1.8.0",
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "-",
    "rssi": -66,
    "bssid": "-"
  },
  "evse": {
    "temp": 28,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 1,
    "loadbl": 2,
    "pwm": 221,
    "solar_stop_timer": 0,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 130,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_sum_mains": 600,
    "solar_max_import": 0,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "modem": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "xxx.xxx.xxx.xxx",
    "port": 1883,
    "topic_prefix": "SmartEVSE-xxxxx",
    "username": "",
    "password_set": 0,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 110,
    "import_active_power": 2.7,
    "total_kwh": 2.5,
    "charged_kwh": 0.9,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 2.5,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 159,
    "L1": 140,
    "L2": 6,
    "L3": 13,
    "last_data_update": 1711131313,
    "charging_L1": true,
    "charging_L2": true,
    "charging_L3": true,
    "original_data": {
      "TOTAL": 159,
      "L1": 140,
      "L2": 6,
      "L3": 13
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}
`

dingo35, 2024-03-22:
You'll have to upgrade to v3.5.0 when it comes out; if you like to test you can use current version v3.5.0-RC2.

Serkri was developed without test abilities for Master/Slave configurations, so you are on uncharted territories ...

vdweegen, 2024-03-22:
I just installed v3.5.0-RC1, so far it works exactly as I was hoping. I'll keep testing. Great work!

vdweegen, 2024-03-22:
Cheers. If I run into any instabilities or bugs I'll be sure to let you know.

dingo35, 2024-03-22:
Great! Please report on the appropriate repo.