serkri/SmartEVSE-3 #256: Communication problem with Fiat 500e (California)

mif11k, 2024-03-28:
Hi,
I used SmartEVSE-2 for several years with no problems (Smart and Solar charging). After I upgraded SmartEVSE to v3 I noticed there is a communication problem with American version of Fiat 500e. The car charges OK, but when charging is stopped regardless of who decides to stop charging (battery at 100% or no sun -> waiting for solar power) the car indicates charging error (two outer charging indicator lights). When the car returns the error and the charger wants to resume the charging process, it is ignored by the car until the door id opened (!!). So far I was not able to reproduce with any other car, that means, other cars can resume charging or when changing the mode Solar / Smart.

Here is the debug output while car reaches 100% at [19. 03. 2024, 7:27:34,350 PM]:

[19. 03. 2024, 7:27:26,978 PM] (I t:2793284ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.9 A IsetBalanced: 20.0 A
[19. 03. 2024, 7:27:27,332 PM] (I t:2793289ms) (printStatus)(C0) L1: 0.9 A L2: 0.7 A L3: 1.5 A Isum: 3.1 A
[19. 03. 2024, 7:27:29,136 PM] (I t:2795291ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.9 A IsetBalanced: 20.0 A
[19. 03. 2024, 7:27:29,357 PM] (I t:2795296ms) (printStatus)(C0) L1: 0.9 A L2: 0.7 A L3: 1.5 A Isum: 3.1 A
[19. 03. 2024, 7:27:30,986 PM] (I t:2797297ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.9 A IsetBalanced: 20.0 A
[19. 03. 2024, 7:27:31,349 PM] (I t:2797302ms) (printStatus)(C0) L1: 0.9 A L2: 0.7 A L3: 1.5 A Isum: 3.1 A
[19. 03. 2024, 7:27:33,045 PM] (I t:2799303ms) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.9 A IsetBalanced: 20.0 A
[19. 03. 2024, 7:27:33,356 PM] (I t:2799308ms) (printStatus)(C0) L1: 0.9 A L2: 0.7 A L3: 1.5 A Isum: 3.1 A
----- car battery full ----
[19. 03. 2024, 7:27:34,350 PM] t:2800642ms) (setState)(C0) 19:27:33 STATE C -> A
[19. 03. 2024, 7:27:35,006 PM] (I t:2801311ms) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.0 A IsetBalanced: 25.0 A
[19. 03. 2024, 7:27:35,384 PM] (I t:2801315ms) (printStatus)(C0) L1: 1.0 A L2: 0.9 A L3: 1.6 A Isum: 3.5 A
[19. 03. 2024, 7:27:36,389 PM] t:2801678ms) (loop)(C1) 0 clients running.
[19. 03. 2024, 7:27:37,006 PM] (I t:2803315ms) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.1 A IsetBalanced: 25.0 A
[19. 03. 2024, 7:27:37,473 PM] (I t:2803321ms) (printStatus)(C0) L1: 1.1 A L2: 1.2 A L3: 1.9 A Isum: 4.2 A
[19. 03. 2024, 7:27:39,068 PM] t:2804143ms) (DisconnectEvent)(C0) EV disconnected for a while. Resetting SoC states(printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.0 A IsetBalanced: 25.0 A
[19. 03. 2024, 7:27:39,418 PM] (I t:2805328ms) (printStatus)(C0) L1: 1.0 A L2: 1.0 A L3: 1.7 A Isum: 3.7 A
[19. 03. 2024, 7:27:41,116 PM] (I t:2807328ms) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.9 A IsetBalanced: 25.0 A
[19. 03. 2024, 7:27:41,538 PM] (I t:2807333ms) (printStatus)(C0) L1: 0.9 A L2: 0.9 A L3: 1.7 A Isum: 3.5 A
[19. 03. 2024, 7:27:43,023 PM] (I t:2809334ms) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.4 A IsetBalanced: 25.0 A

I tested with fw 1.8.0 and 1.7.2.

dingo35, 2024-03-29:
Upgrade to v3.5.0-RC1, that has improved on EV comms.

mif11k, 2024-03-29:
You mean original 3.5.0-RC1? Does it also have Modbus/TCP support? Is there a prebuilt package, or do I have to build myself?

Thanks!

dingo35, 2024-03-29:
No that has been removed because the unsynced modbus calls disturbed the traffic between SmartEVSE's and their modbus connectes kWh meters...

dingo35, 2024-03-29:
Prebuild releases on github can be found on, you guessed it, the releases page:

https://github.com/dingo35/SmartEVSE-3.5/releases

(for some reason hard to find on mobile devices, on desktop it will show at the right side of your screen)

mif11k, 2024-04-08:
@dingo35 I can confirm everything works as expected on 3.5.0-RC2. When charging stops due battery full, charger transitions in "Connected to EV" state. Problem solved.

(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.4 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: 2.4 A L2: 0.0 A L3: 0.2 A Isum: 2.6 A
(D) (Timer100ms)(C0) ModbusRequest 2: Request MainsMeter Measurement
(D) (ModbusSend8)(C0) Sent packet
(D) (ModbusDecode)(C1) Received packet (43 bytes) Response
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=20.5A, ActiveEVSE=1.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.2 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -2.2 A L2: 0.0 A L3: 0.2 A Isum: -2.0 A
(setState)(C1) 14:09:38 STATE C -> B
(setState)(C1) Switching Contactor1 OFF.
(setState)(C1) Switching Contactor2 OFF.
(EVSEStates)(C1) Diode OK
(D) (Timer100ms)(C0) ModbusRequest 2: Request MainsMeter Measurement
(D) (ModbusSend8)(C0) Sent packet
(D) (ModbusDecode)(C1) Received packet (43 bytes) Response
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, IsetBalanced=20.5A, ActiveEVSE=0.
(I) (printStatus)(C0) STATE: B Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.3 A IsetBalanced: 20.0 A
(I) (printStatus)(C0) L1: -6.3 A L2: 0.1 A L3: 0.3 A Isum: -5.9 A

closed 2024-04-08 by mif11k